thenickdude/wacom-driver-fix #2: The pentab is detected but the pen doesn't move the mouse

saiswaroop94, 2020-02-06:
The pentab is detected in the preferences panel and it also allows to change the mapping mode and stuff but the Pentab doesn't move the mouse at all and upon quitting the Pentab driver the Pentab can move the mouse but in mouse mode and not in pen mode

thenickdude, 2020-02-06:
Which tablet model do you have?

saiswaroop94, 2020-02-07:
Thank you for a quick reply I have CTL 470 and all the accessibility settings are turned on properly. Also the input monitoring is checked for pentabletDriver.

thenickdude, 2020-02-08:
Okay, try resetting those accessibility settings like so:

Select PenTabletDriver in the list in the Accessibility tab, and click the minus button to remove it, do the same for Input Monitoring.

Open up the Terminal program and paste these two commands into it (press enter after them to execute them) to reload the tablet driver:

    launchctl unload /Library/LaunchAgents/com.wacom.pentablet.plist
    
    launchctl load /Library/LaunchAgents/com.wacom.pentablet.plist

Then put the pen tip onto the tablet. It should pop up a dialog asking you to allow it access again and re-add it to the Accessibility list for you, tick PenTabletDriver in that list and in the Input Monitoring list, then see if it starts working.

If this doesn't work, run the "Console" program and open up the Crash Reports tab, see if there are any crash reports with Tablet in the name, with today's date on them. If you find one please copy the text of it and paste it into an email to me at n.sherlock@gmail.com

saiswaroop94, 2020-02-11:
Thanks a lot!! the pentab is now working fine!

closed 2020-02-11 by saiswaroop94