thenickdude/wacom-driver-fix #3: after about an hour, my inputs switched

insanepoetry, 2020-02-21:
Thanks for the great fix with the patch. After getting my tablet working again and using zbrush for about an hour i noticed 'option' wasn't working then i noticed my entire keyboard was a different language keyboard, restarted and all good for about another hour again. not sure if this is the right place for this type of thing but you fixed what wacom failed to so.....

thenickdude, 2020-02-21:
The keyboard shortcut for language switching is cmd + [ or ], could you be
hitting this by accident while changing brush size?

insanepoetry, 2020-02-21:
No, keep one hand in the lower left of the keyboard and the other holding the pen. The only keys i would have pressed, even accidentally would pe around the lower left corner, i dont use alot of keys other than shift, cmd, option and space

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/thenickdude/wacom-driver-fix/issues/3?email_source=notifications\u0026email_token=AJYCESVYG57Y5Z7XHGL22X3RD4WM3A5CNFSM4KY2LY52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMRGVRY#issuecomment-589458119",

"url": "https://github.com/thenickdude/wacom-driver-fix/issues/3?email_source=notifications\u0026email_token=AJYCESVYG57Y5Z7XHGL22X3RD4WM3A5CNFSM4KY2LY52YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEMRGVRY#issuecomment-589458119",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

thenickdude, 2020-02-21:
Does your tablet support touch?

insanepoetry, 2020-02-21:
Yes, its a cth-670 i believe, and i checked that part of the permissions, i followed your very well written instructions initially

thenickdude, 2020-02-21:
In your System Preferences, go to Keyboard, then the Shortcuts tab. For "Input Sources", what does it have as the shortcut for "Select the previous/next input source"? I think the default is ⌘[ and ⌘], but if it is assigned to "\^[" or "\^]", then swiping left or right on your trackpad with four fingers would be accidentally switching languages. 

This is because Wacom's four-finger gesture for "switch between full-screen apps" works by typing \^[ and \^] (ctrl + left and ctrl + right). If so, you could fix that by disabling that gesture in Wacom settings.

insanepoetry, 2020-02-21:
Even with touch turned off via the wacom program? And i will check that soon. Mot currently at my computer

thenickdude, 2020-02-21:
Ah, with touch turned off that shouldn't happen! If that's the case then in your system preferences, go to Keyboard, then the Shortcuts tab. On the Input Sources page, untick both of the keyboard shortcuts shown there to disable them. 

Then wait and see if the language-switching problem happens again. This'll help track down exactly what is triggering it.

insanepoetry, 2020-02-21:
Ok, is there something like a log of some kind that i can check as soon as it happens to help narrow it dow ?

thenickdude, 2020-02-21:
macOS does have a system console (the "Console" app), but the log is so incredibly busy that spotting anything in it is difficult (and that's assuming that the cause was being logged in the first place).

The other cause I can imagine for this problem is the system setting Keyboard -> Input Sources -> "Automatically switch to a document's input source", is that ticked for you?

insanepoetry, 2020-02-21:
im a bit confused by your instructions, 'system preferences, go to
Keyboard, then the Shortcuts tab.' what was i supposed to do under this
tab?  in the input sources page, there at no keyboard shortcuts shown. ill
attach screenshots incase ive misunderstood what you wrote.

thenickdude, 2020-02-21:
This page:

[image]

insanepoetry, 2020-02-21:
i dont have that option, mine has screenshots immediately after keyboard

thenickdude, 2020-02-21:
On this page do you have multiple input languages listed? If not, I'm not sure how your system managed to switch to a different keyboard layout:

[image]

insanepoetry, 2020-02-21:
no, i sent you screen shots of that screen and the other one from the last
email. i have been playing with zbrush for a while now and its still
working fine, maybe it was a freak thing although it did happen twice

insanepoetry, 2020-02-21:
It happened again, Just sent you a screenshot of the keyboard input window (i think thats what its called)

[
{

thenickdude, 2020-02-21:
Have you been adding your screenshots as attachments on the GitHub emails you're replying to? GitHub can't accept attachments, in order to include screenshots you need to write your replies in the web interface here:

https://github.com/thenickdude/wacom-driver-fix/issues/3

You can drag and drop screenshots into the textbox that you write in in order to include them.

insanepoetry, 2020-02-21:
ive been replying via gmail or my phone...

insanepoetry, 2020-02-21:
[image]

insanepoetry, 2020-02-21:
i just realized its not a different language, its symbols i believe, still a problem though.

thenickdude, 2020-02-21:
Okay, that's just the display that the accessibility/onscreen keyboard makes if the Alt/Option key is being held down. You can probably clear that state by pressing the Alt key on your keyboard.

In mac's Accessibility -> Keyboard options, do you have Sticky Keys turned on? That would cause Alt/Option to stick down like this:

[image]

insanepoetry, 2020-02-21:
no it is not turned on.  does the faint orange border mean the key is being pressed? i opened that keyboard window once i noticed that alt/option wasnt working correctly in zbrush (how i originaly discovered the problem) in an attempt to illistrate the problem. at the time i took the screen shot (which i had to do by going into utilities-> screenshot becuase cmd+shift+3 wouldnt work) i was not pressing any keys and pressing alt did not make it change back, thats the first thing i tried to do.

thenickdude, 2020-02-24:
Yeah the orange border means that key is pressed.

In your Wacom ExpressKey settings, what actions do you have assigned to each key? If you don't use these buttons, try setting them all to Disabled and see if your problem goes away.

insanepoetry, 2020-03-18:
sorry i forgot about updating you on the situation. It for some reason
hasnt happened since those first few times. I have no idea what cause it
but i haven't had any issues at all lately. Thanks for your help and thanks
for fixing wacom's intentional (as i see it) errors.

closed 2020-03-29 by thenickdude