thenickdude/wacom-driver-fix #4: Intuous 3 on Catalina 10.15.3 Error: There is a problem with your tablet driver

theha, 2020-03-15:
Hi Nick
After follow your instruction, to replace the /Library/PreferencePanes/WacomTablet.prefpane/Contents/MacOS/WacomDriver with your patch and reboot multiple time, I got this error.
Error: There is a problem with your tablet driver
I follow this thread: http://forum.wacom.eu/viewtopic.php?f=6&p=10042
"My ability to find issues with the Intuos 3 driver is very limited because I don't own an Intuos 3 myself. If you open up the "Console" app, check the Crash Reports tab and see if you find anything recent there mentioning your tablet, if so this provides rich information I can use to hunt down and fix bugs in this driver. If not, it's very hard for me to figure out what's going wrong with your tablet."

[WacomTabletDriver_2020-03-15-152506_Thes-MacBook-Air.txt](https://github.com/thenickdude/wacom-driver-fix/files/4335366/WacomTabletDriver_2020-03-15-152506_Thes-MacBook-Air.txt)

thenickdude, 2020-03-16:
Thanks for the crash report!

That's the tablet driver crashing when it tries to load your preferences file. In Finder, click the Go menu, then hold down the Alt key on your keyboard and "Library" should appear, click it, then open the Preferences folder. Give me a copy of the `com.wacom.wacomtablet.prefs` and `com.wacom.wacomtouch.prefs` files you find there. This way I can see if I can replicate the crash on my machine.

After sending me those files, try erasing your preferences to fix the issue. Go to Applications -> Wacom Tablet -> Wacom Tablet Utility, and use it to remove preferences for the Current User and All Users. Then reboot and see if it becomes operational again.

theha, 2020-03-16:
[com.wacom.wacomtablet.prefs.txt](https://github.com/thenickdude/wacom-driver-fix/files/4335863/com.wacom.wacomtablet.prefs.txt)
[com.wacom.wacomtouch.prefs.txt](https://github.com/thenickdude/wacom-driver-fix/files/4335864/com.wacom.wacomtouch.prefs.txt)

theha, 2020-03-16:
Deleting Preferences
Removing /Users/macports/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/macports/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/macports/Library/Preferences/com.wacom.wacomtouch.prefs 
Removing /Users/vincent/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/vincent/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/vincent/Library/Preferences/com.wacom.wacomtouch.prefs 
Removing /Users/stty2u/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/stty2u/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/stty2u/Library/Preferences/com.wacom.wacomtouch.prefs 
Removing /Users/nobody/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/nobody/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/nobody/Library/Preferences/com.wacom.wacomtouch.prefs 
Removing /Users/myguest/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/myguest/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/myguest/Library/Preferences/com.wacom.wacomtouch.prefs 
Removing /Users/alyssa.ny.ca/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/alyssa.ny.ca/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/alyssa.ny.ca/Library/Preferences/com.wacom.wacomtouch.prefs 
Removing /Users/tha/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/tha/Library/Preferences/com.wacom.tabletpreferences 
Removing /Users/tha/Library/Preferences/com.wacom.wacomtouch.prefs 
waiting for WacomTabletDrive to restart.
waiting for WacomTabletDrive to restart.
waiting for WacomTabletDrive to restart.
waiting for WacomTabletDrive to restart.
waiting for WacomTabletDrive to restart.
waiting for WacomTabletDrive to restart.
waiting for WacomTabletDrive to restart.
waiting for WacomTabletDrive to restart.
waiting for WacomTabletDrive to restart.
waiting for WacomTabletDrive to restart.
Done.

theha, 2020-03-16:
After remove BOTH Current user and ALL Users, Reboot 2 times, it works
[image]

theha, 2020-03-16:
Check Prefernces
[image]

theha, 2020-03-16:
And Input Monitoring.
[image]

theha, 2020-03-16:
Many Thanks for your help Nick

thenickdude, 2020-03-16:
Thanks for the comprehensive details! I'm leaving the issue open so that it's easier for people with similar issues to find it, and I'll investigate those preference files now.

thenickdude, 2020-03-19:
Okay it seems like there's a bug in `COnScreenPanelData::ReadSettings()` which causes it to crash if there are more control definitions in the preferences file than it expected to find. 

It looks like your preference file was left-over from a newer Wacom driver version that you had installed, and that newer version has more control definitions in it than this older version was expecting, so it crashes. 

Clearing out those preferences (as you already did) should successfully avoid this problem in the future for you!

closed 2020-03-19 by thenickdude