thenickdude/wacom-driver-fix #5: Intuos 3 PTZ-631W support?

williamlynngray, 2020-03-22:
Hi there, I don't see mention of the Wacom Intuos3 PTZ-631W?  I've got one in great condition but no dice with MacOS Catalina.  Could it be supported please?  Thanks in advance.

thenickdude, 2020-03-22:
Ah, that's one of those oddball tablets that the Wacom US website doesn't even mention the existence of!

It's likely that the regular Intuos 3 driver supports it, so my patched version should work. 

Did you install a too-new driver at any point (like the latest macOS driver from Wacom)? If so, the newer preferences saved by that new driver could be causing the old driver to crash. Go to Applications -> Wacom Utility -> Wacom Tablet Utility, and use it to remove the preferences for the Current User and All Users, then reboot and see if it starts working.

williamlynngray, 2020-03-22:
Presto!  Thanks, that worked, donation sent.

thenickdude, 2020-03-22:
Thanks very much!

I'll see if I can fix my installer to automatically remove these too-new preference files for people in the future to make this easier.

closed 2020-03-22 by thenickdude

thenickdude, 2020-03-22:
Could you please test out this new version of my driver?

https://github.com/thenickdude/wacom-driver-fix/releases/download/patch-5/Install-Wacom-Tablet-6.3.15-3-patched.pkg

Install it over the top of the one you have already, restart when prompted, and then check for me that you can successfully change a tablet setting in the System Preferences and have it survive a reboot.