thenickdude/wacom-driver-fix #7: I can't get it to work for Bamboo CTL-470

viktorstrate, 2020-03-24:
I have tried installing the driver patch, with the automated method as well as the manual method,
but neither work.

After plugging the tablet into the computer. I get this message in the control panel.
I have tried rebooting after install but without luck.

I am running macOS Catalina version 10.15.3.

[image]

> There is a problem with your tablet driver. Please reboot your system. If the problem persists reinstall or update the driver.

emory, 2020-03-24:
heh I just opened an issue somewhat related - i had the same problem but it was because i had the tablet attached via USB and i have the wireless module installed. once i plugged in the usb transceiver and turned on the tablet for wireless use it perked right up. is that something you can try?

viktorstrate, 2020-03-24:
I don't think that my tablet (Bamboo CTL-470) supports wireless mode

thenickdude, 2020-03-24:
If you've ever installed the latest (too-new) Wacom driver, try using Wacom
Utility to remove your preferences and All User preferences and reboot.

viktorstrate, 2020-03-24:
I've just tried to completely uninstall everything and retry. I noticed I get the same error even without connecting my tablet. It first prints:

> A supported tablet was not found on the system.

And after a small amount of time, the previous error message is shown instead.

I looked in Console.app for anything and found a bunch of crash reports.
Here are a few

**Log files**

<details>
<summary>Process name: PenTabletDriver</summary>

```
Process:               PenTabletDriver [20711]
Path:                  /Library/Application Support/Tablet/PenTabletDriver.app/Contents/MacOS/PenTabletDriver
Identifier:            com.wacom.pentablet
Version:               Wacom Tablet 5.3.7-6 (5.3.7f6)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PenTabletDriver [20711]
User ID:               501

Date/Time:             2020-03-24 12:19:51.310 +0100
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Anonymous UUID:        3942E077-AEB5-3C24-19BE-8A31CC1F8A60

Time Awake Since Boot: 10000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000018
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [20711]

VM Regions Near 0x18:
--> 
    __TEXT                 000000010960b000-0000000109ace000 [ 4876K] r-x/rwx SM=COW  /Library/Application Support/Tablet/PenTabletDriver.app/Contents/MacOS/PenTabletDriver

Application Specific Information:
objc_msgSend() selector name: isEqualToString:

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	0x00007fff67e0201d objc_msgSend + 29
1   com.wacom.pentablet           	0x000000010982f671 ProcessUtils::LaunchApplicationWithBundleID(__CFString const*, __CFString const*, unsigned long, bool, __CFString const*) + 335
2   com.wacom.pentablet           	0x0000000109650981 CMacTabletDriver::LaunchTouchDriver() + 51
3   com.wacom.pentablet           	0x0000000109650935 CMacTabletDriver::EstablishCommunicationWithTouchDriver() + 81
4   com.wacom.pentablet           	0x000000010964f43a CMacTabletDriver::Init() + 148
5   com.wacom.pentablet           	0x000000010965ffe8 -[OTabletDriverApplication applicationDidFinishLaunching:] + 23
6   com.apple.CoreFoundation      	0x00007fff31ace35f __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
7   com.apple.CoreFoundation      	0x00007fff31ace2f3 ___CFXRegistrationPost1_block_invoke + 63
8   com.apple.CoreFoundation      	0x00007fff31ace268 _CFXRegistrationPost1 + 372
9   com.apple.CoreFoundation      	0x00007fff31acdebe ___CFXNotificationPost_block_invoke + 97
10  com.apple.CoreFoundation      	0x00007fff31a9d7e2 -[_CFXNotificationRegistrar find:object:observer:enumerator:] + 1575
11  com.apple.CoreFoundation      	0x00007fff31a9cc82 _CFXNotificationPost + 1351
12  com.apple.Foundation          	0x00007fff34122a22 -[NSNotificationCenter postNotificationName:object:userInfo:] + 59
13  com.apple.AppKit              	0x00007fff2eca5fef -[NSApplication _postDidFinishNotification] + 312
14  com.apple.AppKit              	0x00007fff2eca5d32 -[NSApplication _sendFinishLaunchingNotification] + 208
15  com.apple.AppKit              	0x00007fff2eca2df3 -[NSApplication(NSAppleEventHandling) _handleAEOpenEvent:] + 549
16  com.apple.AppKit              	0x00007fff2eca2a39 -[NSApplication(NSAppleEventHandling) _handleCoreEvent:withReplyEvent:] + 688
17  com.apple.Foundation          	0x00007fff3414e246 -[NSAppleEventManager dispatchRawAppleEvent:withRawReply:handlerRefCon:] + 308
18  com.apple.Foundation          	0x00007fff3414e0b0 _NSAppleEventManagerGenericHandler + 98
[... 384 lines cut ...]
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x0000000105224000  rfl: 0x0000000000000200  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x00000000
Trap Number:     0

Binary images description not available

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 1015
    thread_create: 0
    thread_set_state: 0
```

</details>

thenickdude, 2020-03-24:
That's the error generated by the original broken Wacom driver, are you sure you're installing my fix and not the original from Wacom by mistake? When you open my installer the first page explains that this is a fixed version from me, so check for that screen.

viktorstrate, 2020-03-24:
I have tried to install the "Too new drivers" to properly run the uninstaller. And maked sure I am installing your patched version.

I also think that the crash reports are from earlier today, where I might have had an unpatched version.

I found this log from **PenTabletDriver**

`
WACSTATUS CMacHIDCommInterface::OpenInterface(io_object_t, bool) said "kr != kIOReturnSuccess" in /Volumes/Macintosh HD/Users/swalton/Projects/R_5.3.7/Wacom/Mac/Tools/buildbot/R_5.3.7/Wacom/Mac/TabletDriver/Source/Driver/CMacHIDCommInterface.cpp, line 547
`

I don't know if that can have anything to do with it. But I neither have a volume called `Macintosh HD` nor a user called `swalton` on my system. 🤔

I simultaneously got the following logs as well:

**PenTabletDriver (LaunchServices):** `LSExceptions shared instance invalidated for timeout.`
**TabletDriver (LaunchServices):** `LSExceptions shared instance invalidated for timeout.`
**PenTabletDriver (IOHIDLib):** *(ERROR)* `0x100000678: TCC deny IOHIDDeviceOpen`
**PenTabletDriver** `Failed to open Wacom HID Interface [0xe00002e2]`

thenickdude, 2020-03-24:
Follow the directions in the readme to remove and readd all permissions

viktorstrate, 2020-03-24:
Awesome it works now 🎉

I manually added `/Library/Application Support/Tablet/PenTabletDriver.app`, to both `Accessibility` and `Input Monitoring` under `Security & Privacy` -> `Privacy`.

Afterwards I ran the following commands and the Preferences pane loaded correctly after connecting the tablet over USB.

    $ launchctl unload /Library/LaunchAgents/com.wacom.pentablet.plist
    $ launchctl load /Library/LaunchAgents/com.wacom.pentablet.plist

closed 2020-03-24 by thenickdude