thenickdude/wacom-driver-fix #8: New Catalina 10.15.4 broke the patch

luanafaustini, 2020-03-26:
The new release from Catalina broke the patch. The cursor (in my case) stays in the corner of the screen while using the tablet.

thenickdude, 2020-03-26:
Hi there,

Are you using a Bamboo tablet? It's working fine for me on 10.15.4, you may need to either reinstall my driver, or follow the Post-Install Instructions to first remove all the permissions for the driver, reboot, and then re-add them:

https://github.com/thenickdude/wacom-driver-fix/blob/master/Readme.md#post-install-instructions

luanafaustini, 2020-03-27:
Hey! Yes, I am. I had to delete everything and put it all back again. It's working! Thank you :)

closed 2020-03-27 by luanafaustini

halozano, 2020-04-12:
This happened to me when my mac went to sleep and I solved it by unloading and reloading the driver with 

> launchctl unload /Library/LaunchAgents/com.wacom.pentablet.plist
> launchctl load /Library/LaunchAgents/com.wacom.pentablet.plist