thenickdude/wacom-driver-fix #9: Nothing visible on the screen

HuubLel, 2020-03-28:
I have successfully installed the fix and everything seems to work, but I see nothing on the screen.

MacBookPro and Catalina 10.15.4

thenickdude, 2020-03-28:
Screen? Which tablet model do you have? (model name is on the sticker on the back)

HuubLel, 2020-03-28:
Hello,

Thanks for the fast response.

On the sticker it reads:

BAMBOO
Model: CTH-470 (above the barcode is has an additional /K)
S/N: 1 IBP017700

Kind regards,

Huub.

thenickdude, 2020-03-28:
What are you expecting to see on your monitor? Do you mean that the Wacom preferences refuses to open, or...?

HuubLel, 2020-03-28:
Hello again,

My screen is just my desktop, complete.

I can open Wacom Utility App and I can also open Wacom in System preferences and then can do many things. But nothing on my screen.

See attachment.

Kind regards,

Huub.

thenickdude, 2020-03-28:
GitHub doesn't accept attachments by email. You can follow the link to the Issue from the email signature, then drag and drop your image into the textbox on the Web page instead to include it in your message.

HuubLel, 2020-03-28:
May be the utility app is not complete, there is very little for something that is a utiliy, see screenshot. That is all I see.

On 28 Mar 2020, at 12:13, Huub <huub.lelieveld@inter.nl.net> wrote:

Hello again,

My screen is just my desktop, complete.

I can open Wacom Utility App and I can also open Wacom in System preferences and then can do many things. But nothing on my screen.

See attachment.

Kind regards,

Huub.
<Screenshot 2020-03-28 at 12.10.45.jpg>

thenickdude, 2020-03-28:
Wacom Utility only exists to remove preferences and uninstall the driver. The only interface it has is the settings panel in System Preferences.

None of your screenshots came through because you need to use the webpage to add them, not send them by email.

HuubLel, 2020-03-28:
The screenshots
[image]
[image]

thenickdude, 2020-03-28:
Yep that's everything working perfectly normally, that's what the Wacom driver is.

closed 2020-03-28 by thenickdude

HuubLel, 2020-03-28:
I do not understand the answer and also not why this is closed. What is the point in  having this is intrument tf I cannot see anything that I do on the tablet on the screen?

HuubLel, 2020-03-28:
Why closed?

I still cannot see on the screen what I do on the pad.

Closed #9 <https://github.com/thenickdude/wacom-driver-fix/issues/9>.

HuubLel, 2020-03-28:
Normal that I cannot see anything I do on the computer screen?

thenickdude, 2020-03-28:
Using your pen on the tablet should cause the mouse cursor to move on the screen and that's it, that's what a tablet does. What are you expecting to see?

HuubLel, 2020-03-28:
That I would see the contours of the tablet and that I could draw a line, or e.g. a circle around a word in a document. Now it works like a mouse, I don’t need a mouse.

thenickdude, 2020-03-28:
Lots of apps have pen tools for annotating documents (like Preview or Word), and the tablet is just a more natural feeling way of moving the cursor to use these tools.

HuubLel, 2020-03-28:
I am not sure you received my reply. I try again. That attachment is what I want. I tried with Word, Adobe, Preview, Mail, both I never see anything  on the screen. It just works like a mouse.
[image]

HuubLel, 2020-03-28:
... but I never ...

thenickdude, 2020-03-28:
Word provides Inking features for annotations, here is the manual page:

https://support.office.com/en-us/article/draw-and-write-with-ink-in-office-6d76c674-7f4b-414d-b67f-b3ffef6ccf53

HuubLel, 2020-03-29:
Getting closer to understanding, but regrettably:

On the Mac, these features are only available in Office 365 <https://products.office.com/en-US/?ms.url=office365com> and Office 2019 for Mac.

And I have Office 2016.

What about Preview? he manual of Bamboo CTH-470 shows the example “in general “ and states that this can use for many programs. I bought the tablet in 2017 and never used it. When I wanted to see it it did not and I found your programme.  But I still have no use for it. M intendtion is to use it for correcting (editing) articles for reviewing.

Kind regards,

Huub.

https://support.office.com/en-us/article/draw-and-write-with-ink-in-office-6d76c674-7f4b-414d-b67f-b3ffef6ccf53 <https://support.office.com/en-us/article/draw-and-write-with-ink-in-office-6d76c674-7f4b-414d-b67f-b3ffef6ccf53>

thenickdude, 2020-03-29:
Preview does support this, but it's designed for mouse input rather than pen input, so it's difficult to handwrite on it using the pen (it tries to straighten out lines for you, which makes it annoying for handwriting, and this cannot be disabled).

Go View -> Show Markup Toolbar, then click this Sketch button:

[image]

HuubLel, 2020-03-29:
Ok, thanks, I got it working in Preview but indeed annoyingly useless. I though I had bought a useful tool. 

Thanks for answering all of my questions. At least I understand it now.

Kind regards,

Huub.

 <https://user-images.githubusercontent.com/1921411/77836809-56ab1900-71be-11ea-8dec-a299f91bd728.png>

HuubLel, 2020-03-29:
Now I know what to look for, I found that in Acrobat DC reader it works better than in Preview.

 <https://user-images.githubusercontent.com/1921411/77836809-56ab1900-71be-11ea-8dec-a299f91bd728.png>