thenickdude/wacom-driver-fix #10: Menu Icon in Dark Mode doesn't work

allejo, 2020-03-31:
First of all, thank you so much for this patch! 🎉

Now for the actual issue. This isn't a problem with this specific patch. It's been a problem since I can remember. I was wondering if a fix could be incorporated into this repo that fixes the menu icon in dark mode?

**Dark Mode when monitor is active**

[image]

**Dark mode when _another_ monitor is active**

[image]

**Light mode when monitor is active**

[image]

**Light mode when _another_ monitor is active**

[image]

thenickdude, 2020-03-31:
Which driver and tablet is that, is it a Cintiq? There doesn't seem to be any tray icon system for Bamboo tablets (or Intuos 3 as far as I know).

allejo, 2020-03-31:
Tablet: Wacom Bamboo Capture Pen and Touch Tablet (CTH470)
Driver: Install-Wacom-Tablet-5.3.7-6-patched.pkg

thenickdude, 2020-03-31:
Can you show a screenshot of what that icon looks like when you click it? Is there an About menu there which shows a version number or product name? I've never seen this icon before.

allejo, 2020-03-31:
[image]
[image]
[image]

**This icon shows up correctly when a tablet is connected**

[image]

thenickdude, 2020-03-31:
Ahh, I think this is a battery status icon that only appears for tablets with wireless connections, which is why I've never seen it. I'll look into it, but if it's at all difficult then I imagine it won't be worthwhile trying to patch this.

allejo, 2020-03-31:
Oh yea, it's no big deal if this is more trouble than it's worth. This is more so if it's a quick enough fix, that'd be awesome to have the icon fixed. Otherwise, no big deal :smile:

Thanks for looking into this!

thenickdude, 2020-04-01:
[image]

Hmm I tricked the driver into thinking I had a wireless tablet, and the icon actually shows correctly for me on dark mode. Maybe it doesn't work if your computer changes to dark mode after boot? Try rebooting it if so.

allejo, 2020-04-01:
> Edit: Oh does this only happen when the tablet is disconnected? I wasn't able to simulate that here yet

Yea, it only happens when the tablet is disconnected. This behavior has persisted through reboots 😞

thenickdude, 2020-04-01:
My guess here is that because Wacom's "disabled" icon is grey, macOS can't automatically invert the colours of it to match the dark theme. I'm not going to fix this one because it looks like a pain in the ass, but thanks for the suggestion anyway because it was interesting to discover this tray icon existed at all lol.

closed 2020-04-01 by thenickdude

allejo, 2020-04-01:
It's all good, thanks for taking a look!