thenickdude/wacom-driver-fix #13: Mapping for Intuos 3

G-spot-1, 2020-04-12:
All Wacom preferences are working! Except for the Mapping option, which returns an error and closes the pane. Respect.

thenickdude, 2020-04-12:
Can you open up the Console app, go to the Crash Reports tab and see if there is a corresponding crash logged there? If so please send me a copy of the crash report. 

I talked to one person with an Intuos tablet who had a similar problem but it sounded like it was intermittent enough that they were able to make it work - specifically they said that it seemed if they clicked on the mapping button using their mouse instead of their pen it worked.

BETEP-ok, 2020-05-29:
Similar issue with mapping here
Wacom Bamboo pen & touch
CTH-470

In Wacom Settings can not switch to using full surface - click the button and nothing changes. Every second time the settings just crash.
If only central part is used- can map it down no problem, but I want to use the full surface of tha pad, not just the central part.
[image]

thenickdude, 2020-05-29:
Can you open up the Console app, go to the Crash Reports tab and see if there is a corresponding crash logged there? (it may be titled "legacyloader") If so please send me a copy of the crash report to n.sherlock@gmail.com

BETEP-ok, 2020-05-30:
thnx for the fast reply..

experimented today a bit.
the "full surface" radiobutton gives no reaction. just shows the center space of the tablet as seen on the screenshot above.
reducing the interactive area thrower an error as following

in the log I see following events (when filter by "Legacy")

10:46:25.218219+0200	legacyLoader	setLegacyPreferencePanePath: /Library/PreferencePanes/PenTablet.prefpane
стандартное	10:46:39.144361+0200	legacyLoader	*** Assertion failure in -[NSPanel _beginSheet:isCritical:completionHandler:], /AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/ViewBridge/ViewBridge-464.1/NSViewServiceMarshal.m:572
ошибка	10:46:39.152245+0200	legacyLoader	Invalid parameter not satisfying: sheet != self
ошибка	10:46:39.165970+0200	legacyLoader	(
	0   CoreFoundation                      0x00007fff3922bbe7 __exceptionPreprocess + 250
	1   libobjc.A.dylib                     0x00007fff71e6a5bf objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff39254d98 +[NSException raise:format:arguments:] + 88
	3   Foundation                          0x00007fff3b943e9d -[NSAssertionHandler handleFailureInMethod:object:file:lineNumber:description:] + 191
	4   ViewBridge                          0x00007fff6bcd65a5 -[NSWindow(ViewBridgeSwizzle) _beginSheet:isCritical:completionHandler:] + 155
	5   ViewBridge                          0x00007fff6bcd66de -[NSWindow(ViewBridgeSwizzle) swizzledBeginSheet:completionHandler:] + 51
	6   ViewBridge                          0x00007fff6bcd28b1 -[NSViewServiceApplication beginSheet:modalForWindow:modalDelegate:didEndSelector:contextInfo:] + 512
	7   PenTablet                           0x0000000111b60fbd -[OWacomWindowController beginDialog] + 171
	8   PenTablet                           0x0000000111b60ed7 -[OWacomWindowController handleWacomEvent:forViewID:sender:] + 320
	9   PenTablet                           0x0000000111b6130e -[OWacomWindowController dismissDialogWithCode:] + 520
	10  PenTablet                           0x0000000111b61388 -[OWacomWindowController sheetDidEnd:returnCode:contextInfo:] + 20
	11  PenTablet                           0x0000000111b6466b -[OPortionOfTabletController sheetDidEnd:returnCode:contextInfo:] + 39
	12  AppKit                              0x00007fff36cc543a NSWindowEndWindowModalSession + 187
	13  ViewBridge                          0x00007fff6bcd695b -[NSWindow(ViewBridgeSwizzle) swizzledEndSheet:returnCode:] + 46
	14  PenTablet                           0x0000000111b60c9a -[OWacomWindowController okAction:] + 215
	15  AppKit                              0x00007fff36666fc7 -[NSApplication(NSResponder) sendAction:to:from:] + 299
	16  AppKit                              0x00007fff36666e62 -[NSControl sendAction:to:] + 86
	17  AppKit                              0x00007fff36666d94 __26-[NSCell _sendActionFrom:]_block_invoke + 136
	18  AppKit                              0x00007fff36666c96 -[NSCell _sendActionFrom:] + 171
	19  AppKit                              0x00007fff36666bdd -[NSButtonCell _sendActionFrom:] + 96
	20  AppKit                              0x00007fff36662ebb NSControlTrackMouse + 1745
	21  AppKit                              0x00007fff366627c2 -[NSCell trackMouse:inRect:ofView:untilMouseUp:] + 130
	22  AppKit                              0x00007fff36662681 -[NSButtonCell trackMouse:inRect:ofView:untilMouseUp:] + 691
	23  AppKit                              0x00007fff366619fd -[NSControl mouseDown:] + 748
	24  AppKit                              0x00007fff3665fe10 -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:] + 4914
	25  AppKit                              0x00007fff365ca611 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 2612
	26  AppKit                              0x00007fff365c99b9 -[NSWindow(NSEventRouting) sendEvent:] + 349
	27  ViewBridge                          0x00007fff6bcf1dd5 -[NSViewServiceMarshal whileFilteringResponderChain:] + 197
	28  ViewBridge                          0x00007fff6bcea159 -[NSViewServiceMarshal wrap:sendEvent:] + 77
	29  ViewBridge                          0x00007fff6bcea328 -[NSViewServiceMarshal sendEvent:to:] + 230
	30  ViewBridge                          0x00007fff6bcd766d -[NSWindow(ViewBridgeSwizzle) swizzledSendEvent:] + 63
	31  AppKit                              0x00007fff365c7d44 -[NSApplication(NSEvent) sendEvent:] + 352
	32  ViewBridge                          0x00007fff6bcbf711 __65-[NSViewServiceApplication sendEventWithoutCatch:withForwarding:]_block_invoke + 46
	33  ViewBridge                          0x00007fff6bcbf6a6 -[NSViewServiceApplication eventHasNotHitWindow:actions:] + 62
	34  ViewBridge                          0x00007fff6bcbf36a -[NSViewServiceApplication sendEventWithoutCatch:withForwarding:] + 225
	35  ViewBridge                          0x00007fff6bcbf1d7 -[NSViewServiceApplication sendEvent:withForwarding:] + 26
	36  AppKit                              0x00007fff36414daf -[NSApplication run] + 707
	37  AppKit                              0x00007fff363e6b86 NSApplicationMain + 777
	38  libxpc.dylib                        0x00007fff73270984 _xpc_objc_main.cold.3 + 56
	39  libxpc.dylib                        0x00007fff7325f492 _xpc_objc_main + 609
	40  libxpc.dylib                        0x00007fff7325ef93 _xpc_copy_xpcservice_dictionary + 0
	41  ViewBridge                          0x00007fff6bcb5d54 xpc_connection_handler + 0
[... cut ...]
__FONT_DATA                          4K        1 
__LINKEDIT                       389.7M        7 
__OBJC_RO                         32.2M        1 
__OBJC_RW                         1892K        2 
__TEXT                           276.1M      343 
__UNICODE                          564K        1 
mapped file                       73.3M       18 
shared memory                      632K       14 
===========                     =======  ======= 
TOTAL                              1.4G     1133 
TOTAL, minus reserved VM space     1.1G     1133

thenickdude, 2020-05-30:
That central area shown on that display is actually the only active area in the hardware, the tablet doesn't physically support using the pen outside of that region. You can see little crosshairs printed on the tablet's surface which delineate this area.

[image]

BETEP-ok, 2020-05-30:
Feels like you are right. (((

alex-kovac, 2020-09-21:
On Intuos 3 PTZ-631W I was able to access _Mapping_ tab for Pen but each time _Mapping_ tab for mouse was used, the panel would throw an error and close. There was no console crash log created. But, this somehow worked: Selected Pen i Tool chooser, and selected _Mapping_ tab there. Then chose Mouse in Tool choser. It switched to Mouse Mapping settings directly and did not have problems since. 

@thenickdude Thank you for keeping these devices alive!

thenickdude, 2020-09-24:
It looks like this is triggered when the preference pane tries to show a second modal popup when a first one is already open. (The popups that slide down from the top of the window)

@alex-kovac can you try this workaround for me on your Intuos tablet: After the first popup is shown (e.g. the Mapping page), drag the window using the window title, then after that open the second popup (e.g. the "Portion: Set..." button). For me on Bamboo this prevented the crash from happening.

alex-kovac, 2020-09-24:
Hi, @thenickdude. After changing the title this might be a little off-topic. :)
I do not experience issues with `Portion...` popup, `Screen area` or `Tablet area`. It operates as expected. However, the issue I experienced was this:

1. Clicking on 'Wacom Tablet' from 'System preferences', 'Wacom Tablet' panel opens. 'Grip pen' tool is preselected, 'Pen' tab.
2. Clicking on 'Mouse' tool, and then 'Mapping' closes the preferences panel with dialog box: "Preferences error. There was an error in Wacom Tablet preferences."

On my machine, solution for this is to select 'Mapping' tab on 'Grip pen' tool first. Then, click on 'Mouse' tool and enter the 'Mapping' tab this way without an error.

thenickdude, 2020-09-24:
Ah okay, thanks for that detail!

thenickdude, 2020-09-26:
Okay, I think I've tracked this down to a bug in macOS Catalina. I'm going to report the bug to Apple and see what they have to say about it. If they won't fix it then I can likely work around it in the Wacom preference pane instead.

swirly, 2021-04-06:
Hello

First of all MANY thanks for your work. Thanks to you, I can use my tablet instead of having to buy a new one
In big sur, I experience the same problem. Every time I try to activate the preference for the stylet to have only a portion of the screen, the preference pane crash

I have two screens, and I also experience this when I try to map the stylet on second screen : crash the preference pane

I can map the stylet to first screen although

Hoping this can help you to find the bug

ibadenglish, 2022-02-01:
@robertlindin  HALLELUJAH!

thenickdude, 2022-11-24:
For Bamboo tablets, I've now fixed this problem in this release:

https://github.com/thenickdude/wacom-driver-fix/releases/tag/patch-10-bamboo