thenickdude/wacom-driver-fix #15: CTL-470 still giving a 'problem with your tablet' message.

Jhieminga, 2020-04-15:
Installed the patched driver, but when I go to preferences - Wacom I'm still getting a 'there's a problem with your tablet' message. There does seem to be a reaction to using the pen, but the mapping is way off.

thenickdude, 2020-04-15:
Follow the instructions in the post-installation section to remove the
permissions and reboot.

Jhieminga, 2020-04-15:
Yes! That sorted it, thanks! I had removed and reinstalled before, but I guess it didn't reload the drivers the correct way.

closed 2020-04-18 by thenickdude

Jhieminga, 2020-04-21:
I had my tablet working, but when I plugged it in today, it went back to the state it was in before I reloaded the drivers. Do I have to reload them everytime I would like to use my tablet?

thenickdude, 2020-04-21:
No. Did you upgrade macOS to the latest version between then and now? If so, re-do the Post Installation instructions.

Jhieminga, 2020-04-21:
Ok, that works again, but it is a bit of a hassle if this happens everytime my macOS upgrades. I have that set to autoupdate, so I don't always see it when this happens. Hopefully it will continue to work for the next couple of days. And at least I know how to solve the issue now.