thenickdude/wacom-driver-fix #17: Preferences crash persist even after applying fix

LeoTals, 2020-04-25:
Hi, it seems the Intuos 3 bug of crashing the settings didn't go away with the fix. Can you recommend any other steps?

thenickdude, 2020-04-25:
Just to clarify, do you mean the bug where it crashes as soon as you click on anything in the settings menu? Or do you mean that you're getting an error message before the settings even load?

Did you install the automatic installer-based version of the fix? That's the one I'd recommend. 

Also make sure you followed the instructions to remove all the permissions from the Privacy page using the minus button and restart the machine and re-tick the permissions afterwards, this fixes most issues.

LeoTals, 2020-04-25:
The former of the two. I finally got it to work after multiple reinstalls and clearing the perms a good 3 times. Solid work with this patch, thank you.

closed 2020-04-25 by LeoTals