thenickdude/wacom-driver-fix #18: Uninstall?

DrGroovePhd, 2020-04-26:
Hey, I recently did an O/S update that seemed to break this functionality, and then there was a sale...so I bought a new Intuos Pro, and am having an arm wrestle with the security settings in Catalina with the new official driver. Wondering if remnants of this patch are anything to do with it - is there a way to re-set/remove everything and start over?

thenickdude, 2020-04-27:
The uninstaller for the older driver may be unavailable due to being overwritten by the uninstaller for the newer driver. To clean everything up, use Wacom Utility and/or Wacom Desktop Center to uninstall as much of whatever driver you currently have installed as possible. Then install my driver so that you can use the correct older Wacom Utility to remove the old driver completely. Now you have no Wacom drivers installed at all, but permissions will be leftover from both drivers.

In system Security & Privacy settings, go to the Privacy tab and remove anything you can find on the Input Monitoring and Accessibility pages that is remotely related to Wacom (by using the minus button), then restart your computer. Now install the new Wacom driver and everything should be smooth sailing.

DrGroovePhd, 2020-04-27:
Thanks, I did the above and it all seems to be sorted. Loved that you were able to resurrect the tablet in the first place, but (ironically enough given it's not exactly a current model) I rely on it nearly daily for my work. Anyway, I was due to upgrade, purely from a business perspective ;-)

Thanks for your work though. I was glad to send you a very modest donation in appreciation.

thenickdude, 2020-04-27:
Thanks, greatly appreciated!

closed 2020-04-27 by thenickdude