thenickdude/wacom-driver-fix #20: Thanks!, just one thing.

SEMdigital, 2020-05-17:
OK, it WORKS!!!   THANK THANK THANK YOU.

However the .pkg, did not work on 10.15.4 Catalina, it produces a "supported tablet not found".       So I installed the WacomTablet_6.3.15-3 driver from Wacom, I then after the tablet was working, did the Manual install of the two exec files, and it worked great.

Thank you, for the work you do, as soon as I am re-employed, I shall give you a tip.

Thanks!!

thenickdude, 2020-05-18:
Thanks!

Did you do the "remove the old permissions, reboot, and re-add the new" part of the instructions that first time around? You get this sort of symptom if there are still permissions left over from the previous driver.

SEMdigital, 2020-05-18:
I did...  Just didn't work.   The terminal commands to unlauch and relaunch failed as well.     Do a manual restart was called for. 

But, in the end. All is good. 

Thanks

•••

sean@semdi.com
www.SEMDI.com

Sent from my Mobile Command Center. 

This message including any attachments contains valuable business information that is proprietary, privileged, confidential and/or otherwise protected from disclosure. If you are not the addressee or authorized to receive this for the addressee, you must not review, use, disseminate, distribute, print, copy, or take any action based on this message or any information herein. If you received this message in error, please notify the sender immediately of the error by reply e-mail and delete the e-mail and attachment files from your system, retaining no copies in any media. We appreciate your cooperation. S.E. Melnick Digital Imaging.

> 
> 
> Thanks!
> 
> Did you do the "remove the old permissions, reboot, and re-add the new" part of the instructions that first time around? You get this sort of symptom if there are still permissions left over from the previous driver.

-- 
This message including any attachments contains valuable business 
information that is proprietary, privileged, confidential and/or otherwise 
protected from disclosure. If you are not the addressee or authorized to 
receive this for the addressee, you must not review, use, disseminate, 
distribute, print, copy, or take any action based on thismessage or any 
information herein. If you received this message in error, please notify 
the sender immediately of the error by reply e-mail and delete the e-mail 
and attachment files from your system, retaining no copies in any media. We 
appreciate your cooperation. S.E. Melnick Digital Imaging.

kenden, 2020-07-16:
Another report: I also got it working thanks to the fixes, an Intuos 3 PTZ630, on macOS Catalina 10.15.5.

I also had to use the manual instructions from https://github.com/thenickdude/wacom-driver-fix/blob/master/Readme-manual-installation.md, the fixed driver didn't work for me either. Thanks!