thenickdude/wacom-driver-fix #21: Didn't work for me HELP

arralee, 2020-05-18:
So I did all the steps and my Wacom still doesn't align with my screen, the cursor moves but that's what it did usually. Please help, I've seen people say this works

thenickdude, 2020-05-18:
Which tablet model do you have (check the sticker on the back). Are you getting an error when visiting the Wacom Tablet icon in the System Preferences?

arralee, 2020-05-18:
I have a CTL-460
yeah I get this error
[image]

arralee, 2020-05-18:
Wait, I think it just worked

arralee, 2020-05-18:
Anyway thank you so much for making this, it suddenly just worked!!

thenickdude, 2020-05-18:
Magic!

closed 2020-05-18 by thenickdude