thenickdude/wacom-driver-fix #22: 10.15.5 update removed Tablet pref pane

htom1, 2020-05-27:
Hi,
First of all, Thanks for the hard-work and creating the modified drivers! 
I have a CTH-460 and was using the modified drivers & app with 10.15.4 successfully.
Post the 10.15.5 update the tablet preference app (System Preferences) has been removed.
When I first booted into 10.15.5 a warning message was displayed, that the Tablet App is no longer compatible with the current OSX and it has been moved into the incompatible software folder. (Maybe I'm blind but cannot locate the ISF on the HD (root) level....)
Tablet still works, however customisation/ fine tuning is no longer feasible.
Can you take a look? Thanks in advance!

thenickdude, 2020-05-27:
I haven't tested 10.15.5 yet, but last time I updated macOS it did the same thing. It was restored without further troubles by running my driver installer again.

htom1, 2020-05-27:
> I haven't tested 10.15.5 yet, but last time I updated macOS it did the same thing. It was restored without further troubles by running my driver installer again.
Thanks for the speedy response! I'll try the above and revert.

htom1, 2020-05-27:
Yep, running the installer again fixed the issue! Thanks again!

closed 2020-05-27 by htom1