thenickdude/wacom-driver-fix #25: Repeated unloading of driver after wake

alanmoore, 2020-06-09:
First up, this driver is a lifesaver, and has brought my CTH-670 (Bamboo Create medium) back to life in MacOS. Since multi-touch is exclusive to Intuos Pro now, you've saved me the best part of $400. Thank you!

I am running MacOS Mojave 10.14.6 (18G5033) on a 2019 iMac Retina 5K and I rarely reboot. I've noticed that three scenarios can (but don't always) cause the driver to stop working:

1. Restarting the Mac
2. Letting it sleep overnight — especially if I use the quick switch menu to go to the Login Window before walking away. This happens the most consistently.
3. Sometimes it seems to be completely random, in the middle of using the computer

My solution has been to create a shell script to unload and reload the driver:

`#!/bin/bash
echo 'Reloading Wacom Driver';
launchctl unload /Library/LaunchAgents/com.wacom.pentablet.plist;
launchctl load -w /Library/LaunchAgents/com.wacom.pentablet.plist;`  

It's set to run automatically on login, and I have it handy in ~/  so if any of the above three cases happen, I can run the script. It fixes the problem ~80% of the time. ~15% of the time, it will only cause the pen or the touch functions to work, and I run it a second time, which fixes the problem. The rest of the time, it doesn't do anything, and pulling the USB cable out of the tablet and plugging it back in seems to reactivate it.

I wouldn't call this a show stopper, at least for someone savvy enough to follow the instructions in your ReadMe file, and you do mention this fix is specifically for Catalina, which I won't be upgrading to, so if this is too esoteric to track reproduce, or you're simply not going to fix Mojave issues, I completely understand.

Again, thank you for your patch, it's been great to get back to pen and touch after 18 months with a mouse!

thenickdude, 2020-06-09:
You probably want to reload the touch driver too, i.e.:

launchctl unload /Library/LaunchAgents/com.wacom.*

If you open up the Console app are there tablet driver crashes being logged on the Crash Reports page? This might reveal what's going wrong.

alanmoore, 2020-06-09:
There are no crash reports, but I have some speakers connected via USB, and I just powered them up and the tablet stopped working. Ran my script, which got it running again, then took a look in system.log and saw this:

`Jun  9 17:12:57 Alans-iMac TabletDriver[1281]: Failed to open Wacom HID Interface [0xe00002c5]
Jun  9 17:13:04 com.apple.xpc.launchd[1] (com.wacom.pentablet[1281]): Service exited with abnormal code: 255
Jun  9 17:13:05 com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.user.501): Service tried to register a port for an endpoint declared in its launchd.plist(5): service = com.wacom.pentablet, endpoint = Wacom_Driver`

So perhaps there's a conflict with the USB ports?

thenickdude, 2020-06-09:
Do you see other error messages about the USB bus? The speakers could be causing a fault that triggers the USB controller to be reset, and this would be causing havoc with other devices on that controller.

I would suggest this for your script so all the Wacom services are reloaded, not just the one that handles the pen:

```bash
#!/bin/bash 
echo 'Reloading Wacom Driver'
launchctl unload /Library/LaunchAgents/com.wacom.*
launchctl load -w /Library/LaunchAgents/com.wacom.*
```

Nithilher, 2020-06-11:
I can add to this experience.
After wake up of my MacBook Pro from sleep, the tablet does not work properly. As soon as I put the pen on the tablet, the mouse pointer immediately sits in the top left corner of the screen. Moving of the pen has no effect.
However, unloading and reloading of the driver seems to fix things.

alanmoore, 2020-06-29:
Apologies for the late reply, but to follow up, I could not find any USB errors in the various logs exposed by Console, but I have had the USB Speakers cause the tablet to stop working once or twice since. It's not every time though.

Thanks for the updated commands for the reload script, it seems to be working pretty smoothly. Since you are aware of this issue now, would you like me to close this issue?