thenickdude/wacom-driver-fix #26: Not working for me.

aluppi46, 2020-06-19:
Good evening! I have a wacom CTH 461, I use catalina 10.15.5 and I tried several times step by step to install the drive and I couldn't. The image below shows the error it gives. Thank you!
[image]

thenickdude, 2020-06-19:
Hi there,

Did you follow the instructions in this section to first remove all permissions, then reboot and re-tick the entries that reappear?

https://github.com/thenickdude/wacom-driver-fix#fix-up-the-tablet-permissions

This solves most problems. If that didn't fix it, try opening the "Terminal" app, and paste this in and press enter to execute it:

```bash
launchctl load -w /Library/LaunchAgents/com.wacom.*
```

Then reboot your computer and see if it starts working.

aluppi46, 2020-06-19:
Good Morning! Thanks for the quick response. Unfortunately I did all the steps and it didn't work. The same message appears when I click on the wacom icon in the system preferences. The PenTabletDriver, WacomTabletDriver TabletDriver or WacomTabletSpringboard entries do not appear to select, the same goes for the input monitoring. I'm installing the file you provided with the bamboo plugged into the usb, that's how it is. After writing in the terminal launchctl load -w /Library/LaunchAgents/com.wacom.* the image below shows the result, ams still does not work. Anything else I can do?
[image]

[image]
[image]
[image]
[image]

aluppi46, 2020-06-19:
Now it worked!!! Thank you. I opened the Wacom Utility.app in the applications, hit the remove button and the screen below appeared. When clicking on the bamboo with your finger and the pen, the entries appeared to enable them. The screens below follow. Thank you!

[image]
[image]
[image]
[image]
[image]

thenickdude, 2020-06-19:
Ah great, so the problem was caused by the preferences, that's good to know!

closed 2020-06-19 by thenickdude