thenickdude/wacom-driver-fix #27: Support for Graphire Tablets?

DanDawson, 2020-06-22:
Thanks for your work on this! I was curious if there was any chance of it working with the Wacom ET-0405-U Graphire Tablet?

I tried your Graphire3 test release and when I click on the Preference Pane item (Bamboo) I just get "Could not load Bamboo Preference pane."

macOS: 10.15.5

thenickdude, 2020-06-23:
Unfortunately it looks like the Graphire 2 was supported by the "professional" v6 series of drivers, while that experimental Graphire 3 driver I published is from the v5 series, I don't think it supports Graphire 2.

Both Graphire 2 and 3 have the problem that their preference pane was a 32-bit application, so although the 64-bit tablet driver portion might be able to be made to work on Catalina, the preference pane definitely can't be, and would need to be replaced by editing the config files directly.

Could you test out a v6 driver for me? I'll see if I can put something together for it.

DanDawson, 2020-06-23:
I'm glad to test anything you come up with and glad to submit logs or reports on what results I get.

During my initial testing I used multiple versions of Wacom's own drivers, then tried some of yours, and tried to remove the various versions in-between my testing attempts.

thenickdude, 2020-06-23:
Okay, please use Wacom Utility or Tablet Utility to first remove the preferences from All Users, then uninstall the driver, and reboot. 

Now go to the Apple menu, System Preferences, Security and Privacy, the Privacy tab, then from the Accessibility and Input Monitoring pages click the padlock icon to unlock the page, then use the minus button to remove anything even remotely related to Wacom (PenTabletDriver, IOManager etc).

Now install this experimental driver and reboot the computer:

https://github.com/thenickdude/wacom-driver-fix/releases/download/graphire-test-2/Install.Wacom.Tablet-6.1.6-4-patched.pkg

Put the pen tip to the tablet and tick any permissions it asks you to add. The Wacom Tablet preferences will not be available, you'll get an error message when trying to open it. But check to see if the tablet is operating in Tablet Mode: If you put the pen to the top left corner, does the cursor immediately jump there? Or is it operating in Mouse Mode where you have to lift and drag the pen several times to move the cursor around? You might have to reboot one more time for it to start working in Tablet Mode.

If it's still operating in Mouse Mode, or the pen doesn't move the cursor at all, open the "Console" app, and go to the Crash Reports tab on the left. Look for crash reports with today's timestamp that mention the tablet (WacomSpringboard, PenTabletDriver etc), if you find any send me a copy of them to n.sherlock@gmail.com

lesvink, 2020-08-09:
Hello !

Just test your 5-2-6-5 patched driver with my Wacom Graphire 4 (CTE 440) and it works like a charm ! Wouaaa thanks you so much :-)

Two reboot necessary and works !

thenickdude, 2020-08-10:
Okay wow, that's surprising, I didn't realise that driver version supported Graphire 4. Would you be able to test a newer 5.3.0-3 driver version for me too, @lesvink?

lesvink, 2020-08-10:
> Okay wow, that's surprising, I didn't realise that driver version supported Graphire 4, but I see it now in their "older versions" list. Would you be able to test a newer 5.3.0-3 driver version for me too, @lesvink?

@thenickdude Yes it's okay I can test.

thenickdude, 2020-08-10:
One sec, I'm seeing if I can get a preference pane working for this version too...

lesvink, 2020-08-11:
@thenickdude Okay. If you success for the pref pane it will be great :-)

thenickdude, 2020-08-12:
I have the preference pane working now! Please try installing this driver, I'm expecting that it'll cleanly install over the top of the one you have already:

https://github.com/thenickdude/wacom-driver-fix/releases/download/graphire-test-2/Install.Wacom.Tablet-5.3.0-3-patched.pkg

You might have to remove the privacy permissions and reboot if it doesn't immediately work.

lesvink, 2020-08-13:
I test and it's a full success on Catalina 10.15.4. The pref pane works perfectly. Thanks you so much you're great :-D

No need to remove privacy permissions nor reboot. I install and it works immediately. Very good job :-D

[image]

thenickdude, 2020-08-13:
Thanks! I'm hoping to support Graphire 3 too if all goes well.

lesvink, 2020-08-13:
I am sure that you will succeed ;-)

thenickdude, 2020-08-13:
Graphire 3 support has now been published!

By the way I realised that my old convo with @DanDawson in this thread was by private email rather than within this ticket; they reported that my experimental 6.1.6-4 driver did work for their Graphire 2 after enough reboots were applied! I might revisit this again and see if I can finish off the support for that, including the preference pane.

thenickdude, 2020-08-13:
I have a new Graphire 1 and 2 driver available for testing now! The preferences pane now appears to open up, but I'm unable to test it beyond that point myself since I don't have one of these tablets. Happy to hear any test results:

https://github.com/thenickdude/wacom-driver-fix/releases/download/graphire-test-2/Install-Wacom-Tablet-6.1.6-4-patched.pkg

closed 2020-08-28 by thenickdude