thenickdude/wacom-driver-fix #30: I don't get the permission prompts

dapetcu21, 2020-06-25:
Hi! I don't get the permission prompts, with either the unpatched driver or your driver. Additionally, I noticed that your driver doesn't install anything to my `/Library/Extensions`. And if I unpack the `.pkg` with `unpkg.app`, the kernel extensions aren't there. I tried installing the original driver and following the manual patching guide, but I get no permissions prompts either with that method as well. I'm several reboots in. My tablet is a CTE-640.

thenickdude, 2020-06-25:
These Wacom tablets don't need kernel extensions (thankfully!), it's a user-mode program that is installed to /Library/Application Support/Tablet.

Sorry, CTE-640 (Graphire 4) isn't supported by my driver.

closed 2020-06-25 by thenickdude

dapetcu21, 2020-06-25:
Oh. My bad. I confused it with the CTE-460 in your supported list. Thanks

thenickdude, 2020-08-13:
Graphire 4 is now supported by my latest driver release, please give it a try!

dapetcu21, 2020-08-13:
This is awesome. Unfortunately, I don't have the tablet anymore, so can't test it.

thenickdude, 2020-08-13:
Oh well! 🤣