thenickdude/wacom-driver-fix #33: just updated to Catalina 10.15.6 and patch stopped working

danimartmora, 2020-07-24:
Hope you have a chance to fix it

thenickdude, 2020-07-24:
Reinstall my driver (you don't need to uninstall it first), it should fix it

szerge, 2020-08-06:
Hi, after updating to Catalina 10.15.6, I wanted to check my old Bamboo CTH-670/S  (I didn't use for some time). 

I followed the procedure to replace the drivers, unfortunately, the accessibility rights were 'locked' from the previous install. To remove the accessibility rights I removed anything that looked like a Wacom directory or file. Then I installed the pentablet_5.3.7-6.dmg to be in a position follow the procedure from the beginning for the replacement of the drivers. This time accessibility rights question popped-up when I started using the touch board and the pen.

Unfortunately, only the touch pad (and buttons) works, not the pen. And I had 3 items in the accessibility list: com.wacom.tablet.helper (!), ConsumerTouchDriver and PenTabletDriver as expected. I removed  com.wacom.tablet.helper but didn't change the outcome.

Sad as it's still in good shape...

Anyhow, thanks for trying to keep us free from not just 'programmed obsolescence' but 'brut force' one 👍

raffi4, 2020-08-15:
Hi 
thanks for your work 
it look like after 15.5.6 second update Bamboo no longer work ..

the graphire and Intuos has been updated 
but i don't see any news for Bamboo ?
is there any workaround i can  do in order to make it work since few days ago it was working !
regards

raffi4, 2020-08-15:
Thanks 
but at the start up the panel is moved by the system
is it cause i need to run the terminal commands To create the lunch files or anything like this ?
i didn't see you workaround about as today !
regards

thenickdude, 2020-08-15:
By on startup do you mean after installing the upgrade to 10.15.6? Yes, macOS removes it during the upgrade. But then you just need to reinstall my driver afterwards and it comes back. I'm running 10.15.6 here with my Bamboo Pen and it's fine.

patomaki, 2020-08-28:
Hi, thanks for your hard work in debugging and updating the drivers! 

I'm using CTH-661 Bamboo Fun, and indeed had to update to Catalina 10.15.6. After downloading your driver and installing it (after the system update), the Wacom app in System Preferences still states that there is a problem with the driver. Likewise, I never get the prompt for checking the tablet access permissions. But the permissions are already ticked in the Security and Privacy tabs. Is there anything I can do to help you debug?

thenickdude, 2020-08-28:
Remove all the permissions using the minus button and reboot, then re-tick the permissions that reappear when you try to use the tablet.

This solves problems with out of date permissions entries that don't work even when they're ticked.

patomaki, 2020-08-29:
> Remove all the permissions using the minus button and reboot, then re-tick the permissions that reappear when you try to use the tablet.
> 
> This solves problems with out of date permissions entries that don't work even when they're ticked.

Excellent, thanks so much! Sorry I must've missed you advising to do that earlier. That did it, now the tablet is back in action again :)

closed 2020-11-12 by thenickdude