thenickdude/wacom-driver-fix #34: CTF-430 bamboo one, can't set buttons.

nkint, 2020-08-11:
Hi, thanks.
It works, I can see the pen settings in the mac settings:
[image]

But it I click on the buttons drowdopwn nothing happens. Any advice?

thenickdude, 2020-11-22:
Are you still having this issue?

Does the pen work in general? I could see this happening if the pen never entered the tablet's detection zone.

Is your macOS system language in English, or is it some other language?

nkint, 2020-11-23:
Hi @thenickdude thanks to look at this.
yes, the pen in general work without problem.
The system language is English yes... should I try to change it?

thenickdude, 2020-11-23:
English is fine. I'll have a poke around and see if I can figure anything out.

nkint, 2020-11-23:
Tell me if I can send you some other useful information, thanks again!

thenickdude, 2020-11-24:
Can you try using the app "Wacom Utility" to remove your preferences and see if that fixes it?

robertmdm, 2021-04-09:
I had this issue with the Intuos2 on Big Sur and using the Wacom Utility and removing preferences, worked for me <3

gingerbeardman, 2021-08-15:
This also happens to me with the original driver.

SatyrTheTablecloth, 2024-11-18:
@thenickdude I have the same issue with my Bamboo CTH-461 on both my Macs, one running Monterey and the other running Ventura.  At times I can map custom keystrokes, and other times I can't.  I can choose just about every other option, but I bought my tablet so I could set custom keystrokes.  I click on keystroke in the system preferences menu and it does nothing.