thenickdude/wacom-driver-fix #38: Catalina missing Ink System Preferences

leoliberti, 2020-08-28:
Hi. 

I just moved to Catalina from Mojave, and lost the functionality of my Wacom CTH-470 Bamboo tablet. I found your patched driver (impressive work!), and installed it. I fixed the tablet permissions. After that, touching the tablet with my finger prompted the system to install the ConsumerTouchDriver item in "Accessibility" (under Security and Privacy -> Privacy tab); touching it with the pen prompted it to install the PenTabletDriver. 

There is no item at all under "Input Monitoring". There is also no sign of any other item you listed (WacomTouchDriver, WacomTabletDriver, TabletDriver, WacomTabletSpringboard). I can control the pointer with both finger and pen, which means those two drivers are operational. The "Extra-Menu" on the top status line is also active and behaves well. 

I then launched an Ink request with the pen, which spawned a small window warning me that the Ink System Pane should be called up to authorize it. But then, upon pressing the "OK" button, nothing followed. I searched for an "Ink" pane or extra menu everywhere I could (verifying many system locations found online), but no such pane, or extra menu, was to be found anywhere on my system. 

I then found another webpage advertising the fact that Catalina has removed an "Inkwell" component that had been present up to Mojave. I don't know if my missing the "Ink Pane" is due to Catalina removing it from the system, or to something else. 

I think this is likely to be independent of your driver, but perhaps you have a suggestion on how to proceed? Have you encountered this issue before?

Thanks
Yours
Leo Liberti

thenickdude, 2020-08-28:
I've never heard of an ink pane before, it's certainly not part of the
driver or present on Catalina.

You most likely need to restart once before the driver will appear in Input
Monitoring.

The Wacom preference pane is called Wacom Tablet or something like that.
Catalina likes to remove it automatically each time a system update is
installed, so you need to re-run my installer to put it back.

greensouth, 2020-09-08:
Didn't know where to post this, but THANKS! for your efforts. You did what Wacom didn't want to, just to make more money.

Sent you a lil donation 👍

thenickdude, 2020-09-08:
Thanks @greensouth, much appreciated!

closed 2020-09-29 by thenickdude