thenickdude/wacom-driver-fix #39: CTE-630BT on macOS 10.15.6 issue...

PeterLJackson, 2020-08-31:
Thanks for your work on this...

I have a Graphire Bluetooth (CTE-630BT) I am trying to use on a Mac mini (2018) macOS 10.15.6. I installed the driver Install-Wacom-Tablet-5.3.0-3-patched.pkg. The tablet connects via bluetooth as "WACOM Pen Tablet" and is recognized by the Bamboo System Preferences app (the installed driver). The tablet itself registers a a pen touch or motion (the light around the on/off button changes colour), however neither the mouse nor the pen move the cursor on the computer screen. 

If I do not install the driver, then the computer recognizes the bluetooth tablet as a mouse (and can move the cursor, etc.) The problem seems to be related to the Security & Privacy / Accessibility setting:
Under Security & Privacy - Input Monitoring there is a checked "PenTabletSpringboard" entry, but nothing related to the tablet under Accessibility.  I have tried rebooting several times, and also entering the terminal commands unload and load the .plist file as mentioned in the instructions. However nothing appears in the Accessibility list related to the tablet.

Any suggestions on how to proceed?

thenickdude, 2020-08-31:
It sounds like that PenTabletSpringboard permission may be non-functional. 

Remove that PenTabletSpringboard entry by using the minus button, then reboot and try using the tablet again, it should prompt you to re-tick an entry that reappears. You may need one more reboot afterwards for it to prompt for additional permissions.

PeterLJackson, 2020-09-01:
Thank you for your prompt reply. I did as you suggested but could not get the 5.3.0-3 driver to work with my bluetooth Graphire 3 (CTE-630BT). 

However, I found success installing the version of the driver that you listed for the non BT Graphire 3 (5.2.6-5).
So, you may wish to adjust your Readme to specify that the driver for CTE-630BT is the same as the driver for CTE-630 (non BT Graphire 3). (And that this has now been tested!)

Thanks very much for your contribution to this project - I had this tablet sitting in a box unused for several years, and your work saved it from the becoming e-waste.

thenickdude, 2020-09-01:
If you open up the "Console" app and check the Crash Reports section on the left, is there anything there with timestamps in the era of your previous driver install that mentions your tablet? If so I'd love a copy of those crash reports, there may be a fixable problem in the 5.3.0-3 driver (you can send it to n.sherlock@gmail.com).

I'll update the supported driver list for v5.2.6-5 to include the Graphire 3 BT!

thenickdude, 2020-09-01:
Unfortunately I'm a bit baffled with the error report you sent, because it shows that the PenTabletDriver is crashing during loading its UI, and the same driver on my computer is able to run that code successfully. I'm not sure what is triggering the fault here...

You don't have an old app called "Bamboo Dock" installed currently do you? This can't be installed any more on Catalina but maybe you had it left-over on your system from an earlier version of macOS.