thenickdude/wacom-driver-fix #42: Catalina 10.15.7 and patch stopped working

Vyeche, 2020-09-25:
For some reason, I'm not getting the prompts after reinstalling the drivers. 

Modal
CTH-460

Things I tried:
- Removed all remains from accessibility and input monitoring
- Running load and unload

Is anybody else having the issue?

Thank you,
Vye

thenickdude, 2020-09-25:
Try rebooting after removing the permissions (instead of just unload/load)

Vyeche, 2020-09-25:
Hi, @thenickdude, 

I tried your recommendation. The touch functionality works, but the pen and driver in settings are not recognized. Similarly, the accessibility and input monitoring prompts are not asking me to add them. 

Thank you,
Vye

thenickdude, 2020-09-26:
Can you open up the Console app, and in the left column click on Crash Reports? Then on the right side of the window, look for crashes that mention your tablet in the name (or legacyLoader) and have recent timestamps (after your upgrade to 10.15.7). You can send me the reports at n.sherlock@gmail.com

thenickdude, 2020-09-26:
Okay, I've updated to 10.15.7 myself now and I have the opposite problem, it keeps prompting me to add Input Monitoring permissions whenever the driver loads. I'll investigate.

thenickdude, 2020-09-26:
Okay on my system I re-ran my installer to install the fixed driver, then I went to the Accessibility and Input Monitoring settings and removed everything related to Wacom (PenTabletDriver etc), then I rebooted, then when I tried to use the tablet it prompted me to grant Input Monitoring permissions. Then no matter how many times I unloaded and reloaded the driver using the command line it would just ask for Input Monitoring permissions again despite it already being there and ticked.

Then I rebooted macOS again, which fixed it. It stopped asking for Input Monitoring permissions, and instead successfully asked for Accessibility permissions so that it could start moving the mouse cursor for me. Now everything was working.

Vyeche, 2020-09-26:
Hi, @thenickdude, 

Intriguing, maybe my issue lies with something remaining. Is there a way to altogether remove all Wacom components? Error logs? Are you able to remote to my machine?

[image]

[image]

Thank you,
Vye

thenickdude, 2020-09-26:
You can use Wacom Utility / Tablet Utility to delete your preferences and then uninstall the driver.

Vyeche, 2020-09-26:
Hi, @thenickdude, 

I removed it using the utility; then, I reinstalled it. However, something seems to be blocking it.

Thank you,
Vye

thenickdude, 2020-09-26:
What do you mean something's blocking it? What error are you getting and when?

Yep it's a Bamboo tablet that I'm testing.

thenickdude, 2020-09-26:
Can you open up the Console app and go to the Crash Reports page on the left side, and on the right look for crashes that mention the tablet in the name, and that have recent timestamps. You can send me a copy of any crash reports at n.sherlock@gmail.com

Vyeche, 2020-09-28:
Hi, @thenickdude, 

I hope all is well. 

I sent you a few console logs. I don't see any crash logs for the pen tablet. 

Regarding the block, I thought Norton was blocking it, so I tried to disable it, but nothing came of that route.

[image]

Thank you,
Vye

thenickdude, 2020-09-28:
The "TCC deny" error message means that the PenTabletDriver is trying to use Input Monitoring permissions, but is being denied. So the problem your system is having is that even though PenTabletDriver is trying to ask for these permissions, for some reason it is never showing up on the Input Monitoring page for you to tick.

In Finder, click Go -> Go to Folder, and paste this path in there and click OK:

    /Library/Application Support/Tablet/

You should see a "PenTabletDriver" file there. Unlock the Input Monitoring page, then drag the PenTabletDriver file onto it to add it to the list, and make sure it's ticked. Now reboot your computer and hopefully next time you try to use the tablet it'll prompt for you to tick it in the Accessibility page instead, after which it should start working.

Vyeche, 2020-09-28:
Hi, @thenickdude, 

When I try that, the dragged file goes back into the folder and doesn't add to the input monitoring list. 

Thank you,
Vye

thenickdude, 2020-09-28:
Ah unfortunately you've run into a weird macOS snag: If there is nothing at all in the Input Monitoring list, it becomes impossible to manually add items to it.

You need to use some other app which requires Input Monitoring permissions to activate this list for you. It looks like from your earlier screenshots you had "OBS" in the list, if you still have that installed then try having that ask for permissions if possible (probably to enable global shortcut key support).

Vyeche, 2020-09-28:
Hi, @thenickdude, 

Yeah, I even tried with Discord; it didn't ask for it either. 

Thank you,
Vye

thenickdude, 2020-09-28:
I saw someone mention this app as one that will trigger asking for Input Monitoring permissions, you could give that a try:

https://colorsnapper.com/

If that doesn't work then it seems like your entire Input Monitoring system is somehow broken...

Vyeche, 2020-09-29:
Hi, @thenickdude, 

I hope all is well. 

I enabled Input Monitoring by installing the Wacom latest driver. Where is the touch and pen driver for the patched version?

[image]

[image]

[image]

Thank you,
Vye

thenickdude, 2020-09-29:
It looks like you've already got the pen support added. For the touch support you need to open this directory:

    /Library/Application Support/Tablet/PenTabletDriver.app/Contents/Resources/

And drag the ConsumerTouchDriver file into the Accessibility page (it doesn't need to be added to Input Monitoring)

Vyeche, 2020-09-29:
Hi, @thenickdude, 

Thank you, that did the trick. I updated my previous comment. I don't know why we didn't get prompts for the permissions. However, I'm it's resolved. 

Please close the ticket, if there is nothing else.

Vye

thenickdude, 2020-09-29:
Which bit was the key for getting the Input Monitoring to work, was it that Reddit comment about double clicking one of the Wacom apps?

Vyeche, 2020-09-29:
It was the link to download the latest driver; somebody mentioned to try to follow the instructions.

thenickdude, 2020-10-27:
Thanks, I've now updated my readme to include most of these troubleshooting steps.

closed 2020-10-27 by thenickdude