thenickdude/wacom-driver-fix #43: catalina 10.15.7

fieromaury, 2020-10-12:
i receive this error every time i try to load the driver:

Process:               TabletDriver [2011]
Path:                  /Library/Application Support/Tablet/PenTabletDriver.app/Contents/Resources/TabletDriver.app/Contents/MacOS/TabletDriver
Identifier:            com.wacom.TabletDriver
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           TabletDriver [2011]
User ID:               501

Date/Time:             2020-10-12 10:09:13.732 +0200
OS Version:            Mac OS X 10.15.7 (19H2)
Report Version:        12
Bridge OS Version:     4.6 (17P6610)
Anonymous UUID:        FDAAA784-02E0-E308-12EE-08534685417E

Sleep/Wake UUID:       5532B93D-F212-45D9-BEF0-1A3792326238

Time Awake Since Boot: 13000 seconds
Time Since Wake:       3000 seconds

System Integrity Protection: disabled

Crashed Thread:        Unknown

Exception Type:        EXC_CRASH (Code Signature Invalid)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace CODESIGNING, Code 0x1

kernel messages:

Backtrace not available

Unknown thread crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x0000000000000000  rsi: 0x0000000000000000  rbp: 0x0000000000000000  rsp: 0x00007ffee5279b68
   r8: 0x0000000000000000   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x000000011523a000  rfl: 0x0000000000000200  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x00000000
Trap Number:     0

Binary images description not available

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 8433
    thread_create: 0
    thread_set_state: 0

every component has already been added in the security tab, both accessibility and input monitoring...

what should i do?

i have a cth-470/K

thanks
Mauri

thenickdude, 2020-10-12:
That's an error caused by Wacom's broken signature on this driver file. If you use my installer version of the fix it fixes this for you as part of the process, you use that instead. 

The manual installation instructions don't include a fix for this issue because it doesn't actually seem to be required for the Bamboo tablets to function correctly, my guess is that this file is only there to support some older tablet series.

Did you follow the instructions to completely remove all permissions using the minus button and reboot, then re-tick the entries that reappear for you when you try to use the tablet? It fixes most driver issues.

fieromaury, 2020-10-12:
Thankyou fo answering.

this was a fresh install, i removed all the components (after several tries i did) using the utility, than removed with minus and reboot.

than  a clean install of the official driver and a manual install of you patch. Once connected the tablet i did not received any 
kind of message, not even a blinking led on the tablet itself.

i went to privacy settings and realized that (just like any other attempt i did) not any entries was added. i always have to add them manually, as i found in other threads.

Now, after your message i tried to install using your installer on top of my manual attempt, and it seems to have solved that "error" i posted.

Still the tablet is not giving any signs of life.

[image]

this is the message i get in the wacom preferences panel:

[image]

these are the permissions i set (following the paths you gave)
[image]
[image]
[image]
[image]
[image]

thenickdude, 2020-10-12:
What do you get printed to the system log in the Console if you run these two commands in the Terminal?

    launchctl unload /Library/LaunchAgents/com.wacom.*
    launchctl load /Library/LaunchAgents/com.wacom.pentablet.plist

fieromaury, 2020-10-12:
i get this:

[image]

not sure if the last 3 lines refer to this..

thenickdude, 2020-10-12:
Hm, that looks like no errors at all (if Input Monitoring permissions were missing there would be a complaint about TCC Deny). 

Can you try rebooting your computer one more time?

fieromaury, 2020-10-12:
Done. still nothin...

thenickdude, 2020-10-12:
What output do you get if you paste this command into the Terminal?

    ps aux | grep -i tablet

fieromaury, 2020-10-12:
[image]

thenickdude, 2020-10-12:
It looks like the driver is running fine, can you double check that the cable for your tablet isn't broken? Or try a different USB port?

fieromaury, 2020-10-12:
I thought it aswell, but nothin changed...

thenickdude, 2020-10-12:
Click the Apple Menu, then About This Mac, then the System Report... button. On the left, click on USB. Check that your tablet shows up in that list somewhere, it should say CTH-470:

[image]

fieromaury, 2020-10-12:
Not showing up...i'm using a known working usb cable, and usb ports are working perfectly using other devices...

is maybe a fault tablet?

fieromaury, 2020-10-12:
mine are usb c ports, i'm using an adapter but i don't think this could be a problem...

thenickdude, 2020-10-12:
It should show up there even if no driver has been installed, so it seems like you have a broken tablet. Does the tablet's light light up at all?

fieromaury, 2020-10-12:
no light at all..

tried to flip the connector but nothin...tried a different pc, still nothin.
Thanks for the help, at this point it is a broken tablet!

Maurizio

thenickdude, 2020-10-12:
Bummer!

closed 2020-10-12 by thenickdude