thenickdude/wacom-driver-fix #44: CTL 470 not working :( Please Help!

milapraschma, 2020-10-17:
Firstly, I wasn't sure whether we were supposed to uninstall the original Wacom Driver so I did that anyway - should I reinstall it and begin the process again?

Everything looks good and my tablet comes up under the Wacom app in System Preferences - however, it won't work. Trying to paint in PhotoShop and the curser isn't moving. [Had to use the Terminal app to input the commands to give permissions for the tablet. Touching my stylus to the screen didn't prompt the access pop-up.]

I have the CTL-470 Bamboo Pen tablet, and the Intuos LP-190k Stylus. My friend uses this combination on her PC and it worked fine for her. Any suggestions? I have attached screenshots.
 
[image]

[image]

[image]

[image]

thenickdude, 2020-10-27:
It's fine to uninstall the Wacom driver before installing my driver, but you don't need to.

It looks like the driver is fully functional, so all it is missing is the Accessibility permission it needs to move your mouse cursor around. If it already appears in the Accessibility list, use the minus button to remove it and reboot your computer, then re-tick the entry that reappears for you, this should fix it.

If it doesn't appear in the list, go to the directory "/Library/Application Support/Tablet/" in the finder (using Go -> Go to Folder) and drag and drop the PenTabletDriver file you see there into the Accessibility list instead, then reboot your computer if the tablet doesn't start working immediately.

milapraschma, 2020-10-30:
Hmm, still no luck! What can I send through to help? Thank you BTW!

thenickdude, 2020-10-30:
Try holding the Alt key on your keyboard and click the About button that's visible in the bottom right corner of your last screenshot, this will open up a debugging window.

See if it looks like it can detect the pressure and X Y coordinates changing as you use your pen.

milapraschma, 2020-10-30:
[image]

thenickdude, 2020-10-30:
The stylus you're using isn't listed as being compatible with that tablet:

https://machollywood.com/blogs/news/wacom-pen-compatibility

Are you sure that's the combo your friend is using?

thenickdude, 2020-10-30:
Do those numbers in the right column change as you use your pen or not?

milapraschma, 2020-10-30:
Ahh I will double check with her, but no those numbers do not change whatsoever. :(

milapraschma, 2020-10-30:
I am assuming there isn't a way to fix this particular issue short of finding a compatible pen?

thenickdude, 2020-10-30:
If the pen is incompatible this is certainly the symptoms you'd get from it.

Does the LED on the tablet light up brighter when you tap the pen on it?

milapraschma, 2020-10-30:
I am afraid not. I will see if I can find a compatible pen, otherwise I'm assuming this renders my tablet useless?

thenickdude, 2020-10-30:
Right, only specific pen models are compatible, and the others can't be made to work.

milapraschma, 2020-10-30:
Alright, well thank you so much for your help!

closed 2020-10-30 by milapraschma