thenickdude/wacom-driver-fix #46: A supported tablet was not found on the system

kokoshneta, 2020-10-27:
I have an Intuos 3 (PTZ-631W), which has been working fine even under Catalina (10.5.5), though I don’t really use any of the keymapping features, just the pen itself.

I wanted to change some settings, which gave me the preference crash, so I found this patch and installed it. It installed, I gave permissions during the installation, and then it prompted for a restart, which I did.

Having restarted, the pen itself still works as before: I can use it to move the cursor, click, right-click, etc. But when I open the Wacom preference pane, I now get… this:

[image]

Does the installer actually remove the driver for the tablet (and not install a new one in its place)? Or am I missing something?

thenickdude, 2020-10-27:
The installer installs Wacom's full driver, and adds my patch on top.

I haven't actually seen that error before, double check that you installed the correct driver version for your tablet.

kokoshneta, 2020-10-27:
I definitely installed the one called _Fixed driver v6.3.15-3 for Intuos 3 and Cintiq tablets_ (the file `Install-Wacom-Tablet-6.3.15-3-patched.pkg` is still in my Downloads folder), which seems like the only possible option for a PTZ-631W.

thenickdude, 2020-10-27:
Maybe try one more reboot? 

Since your tablet is working fine (I'm assuming it's running in absolute Pen Mode rather than the relative Mouse Mode you get with no working drivers) it means that the driver is running, supports the tablet model, and the permissions are correct, so it's really weird that the preference pane gives this message.

kokoshneta, 2020-10-27:
Nope, rebooting again didn’t fix it either. :-/

The tablet is in Mouse Mode, actually, but it was previously as well – I’ve never gotten used to Pen Mode and prefer to use it in Mouse Mode. So there may well be no working driver.

I think I’ll try reinstalling the patch, just in case, and see if that helps.

kokoshneta, 2020-10-27:
Nope, a second reinstall also didn’t bring back the prefpane’s recognition of the driver. :-(

thenickdude, 2020-10-27:
Okay, in that case definitely try the other troubleshooting steps in the readme, because the driver may not be running. Particularly it won't start at all if it is missing its Input Monitoring permission or that permission is broken.

thenickdude, 2020-10-27:
Oh another thing to try is to use Wacom Utility to delete your preferences, just in case they're wonky.

kokoshneta, 2020-10-27:
Got it!

Removed the existing permissions in Accessibility and Input Monitoring, then rebooted. That prompted re-adding into Input Monitoring. Then rebooted _again_, which prompted re-adding into Accessibility, and _then_ it worked.

I hadn’t noticed the last line in the “Fix up the tablet permissions” section on the help page before, so I probably removed and manually re-added things in between each reboot before – the second reboot was what did it.

Thank you for the super-responsive help!

kokoshneta, 2020-10-27:
Agh, too trigger-happy there.

The prefpane itself works now. In the _Tool_ section, I can now click on _Functions_ and that works fine (all three tabs). But when I click on the “Mapping” tab under _Grip Pen_, I still get the crash that the patch should fix.

Not sure if this is an issue with the patch or something on my machine, but here’s the [crash report](https://github.com/thenickdude/wacom-driver-fix/files/5444785/crash_report.txt) if that tells you anything.

thenickdude, 2020-10-27:
Ah, this issue is a lap reported in #13, but I think this is the first crash report I've seen for this issue, thanks, I'll examine this.

kokoshneta, 2020-10-27:
Yeah, that seems to be close to what’s mentioned in that issue. The only difference is that I don’t have a Mouse in my _Tool_ section, only _Functions_ and _Grip Pen_, and the crash happens when clicking the _Mapping_ tab under _Grip Pen_ tool (unlike alex-kovac, who was able to open the _Mapping_ tab from the _Grip Pen_ tool, just not under the _Mouse_ tool).

thenickdude, 2020-11-12:
Hmm I got an emulated Intuos 5 tablet set up so I could test the preference pane, but I wasn't able to replicate this bug on my machine.

Can you try using Wacom Tablet Utility to backup your preferences, then remove preferences, then try it again? I'm thinking that some problem with the prefs might be the trigger.

kokoshneta, 2020-11-14:
> Can you try using Wacom Tablet Utility to backup your preferences, then remove preferences, then try it again? I'm thinking that some problem with the prefs might be the trigger.

Hmm… not sure whether removing the preferences with the Tablet Utility was the culprit there or something else, but I’m now back to “A supported tablet was not found on the system”. It’s nearly 2:30 in the morning right now, so I’m off to bed, but I’ll have another poke around tomorrow.

When you say, “then try again”, do you mean remove the preferences with the Tablet Utility, then reinstall the driver+patch from the top again? ’Cause I didn’t do that just now, just removed the prefs and then opened System Preferences thinking it would rebuild them automatically if it didn’t find anything.

thenickdude, 2020-11-14:
Nah, by "try again" I just meant to try to open up that Mapping tab in preferences and see if it works now. It will regenerate your preferences for you so you don't need to mess with anything else to have that work.

Not sure what could have triggered the driver to start misbehaving like that, maybe it wants a reboot to make it happy.

kokoshneta, 2020-11-15:
Oh, how utterly stupid of me…

I’d unplugged the tablet (to plug in my flatbed scanner, which I don’t use very often) and completely forgotten about it. Since there was no tablet plugged in, it obviously didn’t know to rebuild the prefs, and “not found on the system” was quite literally true. 🤦‍♂️

Plugged the tablet back in, which immediately prompted the Input Monitoring permission request. Did a reboot (just to be on the safe side), and now everything seems to work, including the Mapping section.

thenickdude, 2020-11-15:
Can you email me a copy of your preferences backup? n.sherlock@gmail.com

closed 2022-11-24 by thenickdude