thenickdude/wacom-driver-fix #47: CTH-460 not working

gianlucapisati, 2020-10-27:
Hi, first of all thanks for your work!

I've got a CTH-460 which doesn't want to work. I see the led flash twice during OSX startup (10.15.6) and then it stays off for the rest of the time...

In the console I can see this log:

```
Oct 27 22:45:02 MBP-gpisati com.apple.xpc.launchd[1] (com.wacom.pentablet[339]): Service exited with abnormal code: 255
Oct 27 22:45:12 MBP-gpisati com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.user.502): Service tried to register a port for an endpoint declared in its launchd.plist(5): service = com.wacom.pentablet, endpoint = Wacom_Driver
```

and this:

```
Process:               WacomTabletDriver [3264]
Path:                  /Applications/Wacom Tablet.localized/*/WacomTabletDriver.app/Contents/MacOS/WacomTabletDriver
Identifier:            WacomTabletDriver
Version:               Wacom Tablet 6.3.41-2 (6.3.41f2)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           WacomTabletDriver [3264]
User ID:               502

Date/Time:             2020-10-27 22:18:14.540 +0100
OS Version:            Mac OS X 10.15.6 (19G2021)
Report Version:        12
Anonymous UUID:        AE6D5951-F9CD-729D-EB29-81631DAE2FB0

Time Awake Since Boot: 200 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: /Library/Frameworks/WacomMultiTouch.framework/Versions/A/WacomMultiTouch
  Referenced from: /Applications/Wacom Tablet.localized/*/WacomTabletDriver.app/Contents/MacOS/WacomTabletDriver
  Reason: image not found

Binary Images:
       0x1093f5000 -        0x109a97fe7 +com.wacom.wacomtablet (Wacom Tablet 6.3.41-2 - 6.3.41f2) <92EE69E7-F100-3E25-A54C-EA54A00C811F> /Applications/Wacom Tablet.localized/*/WacomTabletDriver.app/Contents/MacOS/WacomTabletDriver
       0x113b5b000 -        0x113becf47  dyld (750.6) <F9D4DEDC-8296-3E3F-B517-9C8B89A4C094> /usr/lib/dyld

```

thenickdude, 2020-10-27:
>Version:               Wacom Tablet 6.3.41-2 (6.3.41f2)

This is actually an error from Wacom's latest v6 series driver, which doesn't support your tablet. You'll want to run the uninstaller for that one (from Wacom Desktop Center or Wacom / Tablet Utility, I think), and then afterwards reinstall my driver.

gianlucapisati, 2020-10-28:
Ok, I've uninstalled v6 and I've installed your patched version, but the tablet still doesn't show any life signal. Is it normal? I just see the led blinking twice on startup then it shut down

from the console I just see this log:

```
Oct 28 12:15:54 MBP-gpisati com.apple.xpc.launchd[1] (com.wacom.pentablet[354]): Service exited with abnormal code: 255
Oct 28 12:16:04 MBP-gpisati com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.user.502): Service tried to register a port for an endpoint declared in its launchd.plist(5): service = com.wacom.pentablet, endpoint = Wacom_Driver
```
but maybe I'm looking in the wrong place. What should I check to understand what's going wrong?

thenickdude, 2020-10-28:
That last error is triggered by more leftover v6 stuff.

Please try the instructions in the readme to remove all the permissions and reboot.

closed 2020-11-21 by thenickdude