thenickdude/wacom-driver-fix #48: pressure input does not work with Illustrator (but it does with Photoshop)

soklamon, 2020-10-29:
As the title indicates, the pressure input works fine in Photoshop (like with the brush tool) but it does not work in Illustrator. The option of pressure is greyed out and cannot be activated. 
Any suggestions?

thenickdude, 2020-10-29:
If this is a problem with the driver, it's one that can't practically be fixed, sorry.

Could you be selecting a brush that doesn't support pressure? I'm not familiar with Illustrator. I found this series of screenshots showing how to activate it:

http://www.sunnysidesoft.com/virtualtablet/photoshop-pressure-support/

closed 2020-10-29 by thenickdude

soklamon, 2020-10-29:
It works in Photoshop but not in Illustrator. any idea why it does selectivelly. Thanx for the screenshots they show the basic setup of the pressure, as said it is grayed out.