thenickdude/wacom-driver-fix #50: Just an info ctl-470 (driver v5)

GiulioBen, 2020-11-06:
Thank you for your great work!!

now my old bamboo is working again.

However I have 2 monitors and I'm no more able to map the tablet in the second one... at the moment it covers both the screens...

Do you have any tip to allow me to select only the second screen? I tried editing the xml config file but with no results...

thank you again!

thenickdude, 2020-11-07:
It looks like the Mapping settings in Wacom Preferences are a little touchy. First click All Screens (even if it already appears selected) then click OK:

[image]

Then reenter the Mapping settings and you should be able to select Monitor and set it to 2:

[image]

It looks like there is a bug where if the mapping is currently set to a monitor, like monitor 1, attempting to directly change it to monitor 2 causes the preferences to crash, but doing it like this instead by selecting All Screens first avoids the issue.

GiulioBen, 2020-11-07:
But is the wacom setting available in preferencss with Catalina?

I can't fint id anymore since the update!

thenickdude, 2020-11-07:
If you installed a system update, they tend to remove the preference pane automatically as Incompatible Software (I have a fix for this queued up at the moment). You need to run the installer again to bring it back.

closed 2020-11-07 by thenickdude