thenickdude/wacom-driver-fix #51: Request: Install from Homebrew

TheElectroclassics, 2020-11-14:
Hey! I've been using your driver for the last 6 months and it works amazing! Now that I've updated to Big Sur, going to need to reinstall it, so I was thinking, is there a way to get the installer on to homebrew to make the update easier?

thenickdude, 2020-11-14:
You provably reinstalled because it removed the Preference Pane during the upgrade as Incompatible Software, right? I worked around that in this new release so that shouldn't happen again. Maybe that will obviate the need for a more convenient installer?

TheElectroclassics, 2020-11-14:
Yes, and Yes, LOL.

I mean, the fix is so reliable that I forget about it until I have to update and macOS deletes the pref panel and I get a panic attack trying to remember the name of your repo.

closed 2020-11-16 by thenickdude