thenickdude/wacom-driver-fix #52: Driver unload after restart (Intuos 3)

headsign, 2020-11-16:
Hello. The installer works like a charm on my Mac Pro 5,1 running Catalina via OpenCore. 
Unfortunately, the driver unloads after every restart and I have to reinstall for the current session. Any ideas?

thenickdude, 2020-11-16:
Hmm it might be crashing if it is loaded during startup. Open up the Console app and go to the Crash Reports page on the left, and check if you're accumulating crashes that mention the tablet in the name.

You can at least avoid having to reinstall by reloading the driver by pasting this into the Terminal to restart the driver:

    launchctl unload /Library/LaunchAgents/com.wacom.wacomtablet.plist
    launchctl load -w /Library/LaunchAgents/com.wacom.wacomtablet.plist 

Actually, pasting this into the terminal will also permanently fix a possible cause of your symptoms by enabling the Wacom service to load on startup if it was disabled.

headsign, 2020-11-16:
Hmm, I checked the console but the driver didn't crash, apparently. Also, had already tried the 'unload and load', as described in another thread. No luck.

thenickdude, 2020-11-16:
Did you try that load command with the -w included in it? That's what makes the load persistent after a reboot. I didn't include that in some of my earlier Issues.

headsign, 2020-11-16:
Yes, I did include -w.

thenickdude, 2020-11-16:
What's the output of running this in the Terminal after a reboot:

    ps aux | grep -i tablet

headsign, 2020-11-16:
xxxxx     722   0.0  0.1  4899084  12256   ??  S    11:21AM   0:00.13 /Library/Application Support/Tablet/WacomTabletDriver.app/Contents/Resources/WacomTouchDriver.app/Contents/MacOS/WacomTouchDriver
xxxxx      662   0.0  0.3  4450944  56516   ??  S    11:21AM   0:00.93 /Library/Application Support/Tablet/WacomTabletDriver.app/Contents/MacOS/WacomTabletDriver
xxxxx     1528   0.0  0.0  4277516    716 s000  S+   11:25AM   0:00.00 grep -i tablet
xxxxx      778   0.0  0.1  4897712  11616   ??  S    11:21AM   0:00.16 /Library/Application Support/Tablet/WacomTabletDriver.app/Contents/Resources/TabletDriver.app/Contents/MacOS/TabletDriver

thenickdude, 2020-11-16:
Intriguing, the tablet driver processes are all running... and no crashes are logged.

If the tablet driver gets denied Input Monitoring then this is the symptom you'd get, but I can't imagine why that would be denied at startup but work again later on. I haven't seen this before. 

Are you running any special apps that monitor or gatekeep permissions?

headsign, 2020-11-16:
No special gatekeeping apps.

thenickdude, 2020-11-16:
I don't suppose that unplugging and replugging the tablet is enough to kick it into life?

headsign, 2020-11-16:
No. Unplugging and replugging didn't work.
The exact same setup works fine under Mojave, btw. I transferred the system settings from there and reinstalled the driver.

headsign, 2020-11-16:
Don't know if this is in any way relevant, but I doubleclicked WacomTabletSpringboard and this is what came up:

xxxxx ~ % /Library/Application\ Support/Tablet/WacomTabletSpringboard ; exit;
2020-11-16 12:03:12.727 WacomTabletDriver[8755:68320] *** CFMessagePort: bootstrap_register(): failed 1100 (0x44c) 'Permission denied', port = 0x6ea7, name = 'com.wacom.WacomTabletDriverMessageServer'
See /usr/include/servers/bootstrap_defs.h for the error codes.
2020-11-16 12:03:13.728 WacomTabletDriver[8755:68320] *** CFMessagePort: bootstrap_register(): failed 1100 (0x44c) 'Permission denied', port = 0x6eab, name = 'com.wacom.WacomTabletDriverMessageServer'
See /usr/include/servers/bootstrap_defs.h for the error codes.
2020-11-16 12:03:14.730 WacomTabletDriver[8755:68320] *** CFMessagePort: bootstrap_register(): failed 1100 (0x44c) 'Permission denied', port = 0x6eb3, name = 'com.wacom.WacomTabletDriverMessageServer'
See /usr/include/servers/bootstrap_defs.h for the error codes.
Nov 16 12:03:15  TabletDriver[8755] <Error>: FAILED to Create Messaging Server for CPL

[Process completed]

thenickdude, 2020-11-16:
Those are expected I think, it's not supposed to be launched like that.

Did you try removing the Input Monitoring permission with the minus button and rebooting?

headsign, 2020-11-16:
Oh, wow! That did the trick. Thanks for support!

closed 2020-11-16 by headsign