thenickdude/wacom-driver-fix #53: Big Sur loads the tablets but doesn't register pen input.

TheElectroclassics, 2020-11-16:
So, I have a CTL-460 and a CTH-470. macOS loads the preferences and reads the tablets but fails to register pen input. The tablets do register the inputs but not the system. On the 470 (Touch-enabled Bamboo tablet) the system registers touch input too.

thenickdude, 2020-11-16:
When you say the tablet registers the inputs, do they appear in the Wacom tablet preferences if you hold the alt key and click the About button to bring up the tablet diagnostics page? Normally you can see the X and Y coordinates and the pressure changing here as you move the pen.

isaydesign, 2020-11-18:
I have a Touch-enabled Bamboo tablet (CTH-470/K) and everything's fine on Big Sur except for touch input, it doesn't register. Using a tool like Affinity Designer for digital painting, I lost the hability to rotate and zoom the canvas with my fingers, which is super useful.

isaydesign, 2020-11-18:
**Update:** turns out it does register, but nothing happens. I get readings on all types of touches, and the 3&4 finger gestures work. **BUT** the cursor doesn't move, and although the icons that indicate zoom and rotation appear, the rotation and zoom don't actually occur.
[image]

thenickdude, 2020-11-20:
@isaydesign I can replicate your problem here now too. 

My Bamboo touch support was working perfectly after upgrading to Big Sur, but when I removed ConsumerTouchDriver's permission from the Accessibility section, re-adding it and rebooting didn't bring it back to working again.

I'll see if I can figure this out...

isaydesign, 2020-11-20:
@thenickdude thanks in advance for your effort.

thenickdude, 2020-11-21:
@isaydesign please try out this new version for me, I believe it fixes touch support on Big Sur now:

https://github.com/thenickdude/wacom-driver-fix/releases/tag/patch-8

isaydesign, 2020-11-21:
@thenickdude just to let you know it works perfectly. Thanks!

m-thld, 2021-09-02:
Hi there @thenickdude , and thanks for your patch ! I went from no tablet working to a tablet that Big Sur finally reckon, but only with the touch ! Just like @TheElectroclassics. In the about section the X and Y is moving when using the pen, but nothing happens on the screen.
Any idea how to fix this ?