thenickdude/wacom-driver-fix #54: Intuos3 PTZ-630 doesn't work in clear install Big Sur

superlativo77, 2020-11-30:
First, great work @thenickdude and thank you very much for working against planned obsolescence and consumerism.

I have a PTZ-630 model and I installed Install-Wacom-Tablet-6.3.15-3-patched.pkg without problems but it doesn't detect it.

I am attaching screenshots of the two "Security and Privacy" sections.

Any ideas?

Thanks in advance.
[image]
[image]
[image]

thenickdude, 2020-11-30:
That "com.wacom.IOManager" shown in your second screenshot is a leftover from Wacom's newest driver, which doesn't support your tablet. You'll want to run the uninstaller for that one (possibly through "Wacom Utility") to remove it before reinstalling my driver.

eduardoliron, 2021-01-28:
Same issue here, with PTZ-630

[image]
[image]
[image]

thenickdude, 2021-01-29:
@eduardoliron which of the troubleshooting instructions in the readme did you follow so far?

eduardoliron, 2021-01-29:
I´ve followed it correctly.

Turns out that after a couple restarts, the driver started working as expected.

I thank you very much for your work.

eduardoliron, 2021-01-29:
ps: Thr WacomTabletSpringboard permission in the accessibilty pane was the permission that was missing, as I could understand.

It was only asked to me after two or tree restarts though.

[image]