thenickdude/wacom-driver-fix #55: There is a problem with your tablet driver, please reboot your system

wilttd, 2020-12-05:
removed all bamboo drivers and programs

installed:
My fixed Bamboo driver (v5.3.7-6) supports these tablets:

for tablet/mac:
CTH-661/s - Bamboo Fun
iMac Late 2013, macOS Catalina

The patched driver does not crash the OS while booting, works with your fingers. but not with the pen.

in the Wacom app in the system preferences i get the error:
There is a problem with your tablet driver, please reboot your system

wilttd, 2020-12-05:
error before the there is a problem with your tablet driver is there is no tablet connected to your computer

ellejaysee, 2020-12-05:
Same issue here. I removed the security & privacy items you listed but still see "PenTabletSpringboard". I'm not a coder, but an designer who is trying to figure out how to use this tablet again. Never had issues before. Your help is appreciated.
[image]

thenickdude, 2020-12-05:
@ellejaysee "PenTabletSpringboard" isn't a feature of my Bamboo driver, do you have another driver installed or is your tablet not a Bamboo tablet? In any case, don't untick that entry, use the minus button to remove it completely as the readme instructs.

thenickdude, 2020-12-05:
@wilttd did you follow the instructions to remove all the tablet permissions from Input Monitoring and Accessibility and reboot?

ellejaysee, 2020-12-05:
@thenickdude  ok so I just deleted that item and restarted. I see this when I reboot. Do I need to do anything with this? 
Regarding @wilttd's question, I deleted the tablet permissions from Input Monitoring and Accessibility sections and encountered the same thing as he. Hoping this is an easy fix!
[image]

thenickdude, 2020-12-05:
Please follow the instructions in the readme, you're at this section now:

https://github.com/thenickdude/wacom-driver-fix/blob/master/Readme.md#fix-up-the-tablet-permissions

ellejaysee, 2020-12-05:
That worked!!!! Thank you!!

wilttd, 2020-12-12:
Yes i did remove all permissions before installing the patched driver, even did the command line removal but no success.

a2nl2e, 2022-08-25:
I was able to use the patch for about 10 days and then it stopped. I'm running catalina Intous3. I've uninstalled and reinstalled, but I can seem to drag springboard into Input Monitoring. Help please !