thenickdude/wacom-driver-fix #56: How can I completely remove this driver?

antonkratz, 2020-12-09:
I installed this driver a few months back - wanted to use my Wacom Bamboo again and could - I am sorry, I cannot reconstruct anymore exactly which steps I took - now I am on macOS Big Sur 11.01 - I have difficulties with Illustrator 2021 and believe the Wacom driver(s) might be involved - I just want to COMPLETELY remove these drivers again - including the thing in the upper right corner (little Wacom icon ("No Tablets Were Found" /  Preferences).

Please I just want to entirely remove this from my system again.

How can I do that?

thenickdude, 2020-12-09:
Use the app "Wacom Utility" (or Tablet Utility), it provides an Uninstall option

antonkratz, 2020-12-09:
> Use the app "Wacom Utility" (or Tablet Utility), it provides an Uninstall option

I already did that, but the icon in the upper right still persists, even after multiple reboots!

I also deleted Wacom Utility and Wacom Tablet from the Applications folder, but as the icon persists, I assume there must still be remnants of the driver.

thenickdude, 2020-12-09:
That preferences icon is from Wacom's v6 series driver, not my v5 driver for Bamboo. You may have one more Wacom Utility still there that corresponds to that driver instead.

thenickdude, 2020-12-09:
Click the Apple menu then About This Mac, click the System Report... button, then on the left select Software > Applications and wait for the page to load (takes about 10 seconds for me). Check which numbers are in the "Version" column for "com.wacom" (e.g. 6.3.41-2).

antonkratz, 2020-12-09:
com.wacom.DataStoreMgr 1.0
com.wacom.TabletHelper 1.0
com.wacom.UpdateHelper 1.0
PenTabletDriver Wacom Tablet 5.3.7-6
WacomTabletDriver 6.3.29-6

thenickdude, 2020-12-09:
Okay 6.3.29-6 is the official Wacom driver version, you can remove that by re-installing Wacom's official driver, which will bring back Wacom Utility for you, then you can use Wacom Utility to remove it completely. Here's the download link for that version:

https://cdn.wacom.com/u/productsupport/drivers/mac/professional/WacomTablet_6.3.29-6.dmg

**After** you've done that, you can manually remove PenTabletDriver 5.3.7-6 using Finder. Click Go -> Go to Folder, then enter this and hit Okay:

    /Library/Application Support

In that folder select the folder labelled "Tablet" and delete it (enter your login password to confirm).

antonkratz, 2020-12-09:
Thank you!!! I followed your instructions. The small icon in the upper right is gone! I am able to use Illustrator 2021 again, which seems to confirm that there was an interplay between some of the drivers and Illustrator 2021. The only remaining thing is that I still have the Wacom Tablet icon in the Preferences, which confuses me, because I thought this is what I just deleted. Just wondering if I could have a comment on how to get rid of that one as well?

closed 2020-12-09 by antonkratz