thenickdude/wacom-driver-fix #58: CTE-430

gosugimoto, 2020-12-19:
Who would've thought I could maybe start using this again.  I tried to install the driver, but I get a "A supported tablet was not found on the system."

I can use it and the computer responds, but not in the correct way - not mapping the tablet area to screen or anything else.. Would love to use it, I'm on mac 10.15.6 catalina

Thanks!!!

thenickdude, 2020-12-19:
Did you follow the instructions to fix up your permissions?

gosugimoto, 2020-12-19:
Yea and did all the settings in the system preferences.  :(

thenickdude, 2020-12-19:
Did the entries appear in Input Monitoring and Accessibility for you successfully?

gosugimoto, 2020-12-19:
I had to do some manila things as I found in one of the solutions in the
issues or Reddit pages...but that didn't work for me either. Perhaps need
to restart again from scratch?

thenickdude, 2020-12-19:
Yeah if you remove all the permissions from Input Monitoring and Accessibility and restart, hopefully it will pop up and ask you to tick Input Monitoring for PenTabletSpringboard. Then after ticking that you need one more restart before it will ask for Accessibility.

If it never appears in Input Monitoring, try open up the "Terminal" app, and paste this line in there and press enter:

    launchctl load -w /Library/LaunchAgents/com.wacom.pentablet.plist

That may make it fire into life.

gosugimoto, 2020-12-21:
Not sure what it was, but tried a re-install and now working!! Thank you so
much for reviving an old tech I would've otherwise pawned off for a few
quid.

Thanks!!!
--
iamgo.net

thenickdude, 2020-12-21:
No problem!

closed 2020-12-21 by thenickdude