thenickdude/wacom-driver-fix #60: Bamboo fun CTE 650W doesn't work properly

alaghi000, 2021-01-26:
It works for a while, and suddenly it stops to work and need to restart to come back to work.

thenickdude, 2021-01-26:
Open the "Console" app, and on the left side select "Crash Reports", and see if there are any recent crashes logged that mention your tablet or PenTabletDriver in the name.

alaghi000, 2021-01-26:
`Process:               PenTabletDriver [1170]
Path:                  /Library/Application Support/Tablet/PenTabletDriver.app/Contents/MacOS/PenTabletDriver
Identifier:            com.wacom.pentablet
Version:               Wacom Tablet 5.3.7-6 (5.3.7f6)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PenTabletDriver [1170]
User ID:               501

Date/Time:             2021-01-26 15:43:48.464 -0300
OS Version:            Mac OS X 10.15.7 (19H2)
Report Version:        12
Anonymous UUID:        8C36BDB3-0E60-B4B4-14BB-A69EE3493BFD

Time Awake Since Boot: 3200 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [1170]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_malloc.dylib        	0x00007fff683ad8b4 tiny_malloc_from_free_list + 162
1   libsystem_malloc.dylib        	0x00007fff683ad297 tiny_malloc_should_clear + 288
2   libsystem_malloc.dylib        	0x00007fff683ac0c6 szone_malloc_should_clear + 66
3   libsystem_malloc.dylib        	0x00007fff683aad7a malloc_zone_malloc + 104
4   libsystem_malloc.dylib        	0x00007fff683aacf5 malloc + 21
5   libc++abi.dylib               	0x00007fff65523dea operator new(unsigned long) + 26
6   com.apple.analyticsd          	0x00007fff45a88fca std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string<std::nullptr_t>(char const*) + 90
7   com.apple.analyticsd          	0x00007fff45a89509 FrameworkConfiguration::searchWhitelist(std::__1::basic_string_view<char, std::__1::char_traits<char> >) const + 121
8   com.apple.analyticsd          	0x00007fff45a89486 FrameworkConfiguration::isEventWhitelisted(std::__1::basic_string_view<char, std::__1::char_traits<char> >) const + 18
9   com.apple.analyticsd          	0x00007fff45a86e98 CoreAnalytics::Client::isEventWhitelisted(std::__1::basic_string_view<char, std::__1::char_traits<char> >) const + 54
10  libDiagnosticMessagesClient.dylib	0x00007fff64d04762 msgtracer_domain_new + 1117
11  libDiagnosticMessagesClient.dylib	0x00007fff64d0506a msgtracer_vlog_with_keys_skip_nulls + 154
12  libDiagnosticMessagesClient.dylib	0x00007fff64d04fb2 msgtracer_log_with_keys + 126
13  com.apple.framework.IOKit     	0x00007fff310a71e4 IOHIDPostEvent + 196
14  com.apple.SkyLight            	0x00007fff5d488081 SLSUpdateSystemActivityWithLocation + 126
15  com.apple.SkyLight            	0x00007fff5d4880fa CGSTickleActivityMonitor + 45
16  com.wacom.pentablet           	0x000000010cbfa3e5 CMacHIDGestureEventOSX1010::PostGesture(EIOHIDEventType, float) + 199
17  com.wacom.pentablet           	0x000000010cd7ab67 CMacCommandPublisher::PostApplicationCommand(CAppGestureCommand const&, double) + 351
18  com.wacom.pentablet           	0x000000010cc116c1 CCommandPublisher::PostApplicationCommandPrivate(CAppGestureCommand const&, SAccumulator&) + 181
19  com.wacom.pentablet           	0x000000010cc113f1 CCommandPublisher::ProcessZoom(double, bool) + 225
20  com.wacom.pentablet           	0x000000010cd094ac CWinInterface::ProcessTagListZoom(CTagList*) + 182
21  com.wacom.pentablet           	0x000000010cd080b1 CWinInterface::ProcessCurrentTagList(CTagList*) + 1777
22  com.wacom.pentablet           	0x000000010cb1c6a5 CInterfaceManager::ProcessWaitingTagList(CTagList*) + 165
23  com.wacom.pentablet           	0x000000010cb1c58b CInterfaceManager::ProcessCurrentTagList(CTagList*&) + 303
24  com.wacom.pentablet           	0x000000010cb35ebd CGraphicsTablet::PostProcessCurrentTagList(CTagList*&) + 517
25  com.wacom.pentablet           	0x000000010ccc8ce8 CTabletControls::ProcessTabletControls(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > const&, unsigned short) + 132
26  com.wacom.pentablet           	0x000000010cc2d6ac CCTEUSBGraphicsTablet::ProcessTabletControls(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > const&) + 84
27  com.wacom.pentablet           	0x000000010cc2d5e7 CCTEUSBGraphicsTablet::ProcessDataPacket(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > const&) + 243
28  com.wacom.pentablet           	0x000000010cc1b6e4 CCommInterface::TriggerReadCallback(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > const&) + 22
29  com.wacom.pentablet           	0x000000010cb59d8d CMacHIDCommInterface::ProcessDataSynchronously(int, unsigned int) + 413
30  com.apple.iokit.IOHIDLib      	0x000000010ef71a92 0x10ef64000 + 55954
31  com.apple.CoreFoundation      	0x00007fff2e34cb2b __CFMachPortPerform + 288
32  com.apple.CoreFoundation      	0x00007fff2e31e304 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 41
33  com.apple.CoreFoundation      	0x00007fff2e31e250 __CFRunLoopDoSource1 + 541
34  com.apple.CoreFoundation      	0x00007fff2e31cd79 __CFRunLoopRun + 2270
35  com.apple.CoreFoundation      	0x00007fff2e31be3e CFRunLoopRunSpecific + 462
36  com.apple.HIToolbox           	0x00007fff2cf48abd RunCurrentEventLoopInMode + 292
37  com.apple.HIToolbox           	0x00007fff2cf487d5 ReceiveNextEventCommon + 584
38  com.apple.HIToolbox           	0x00007fff2cf48579 _BlockUntilNextEventMatchingListInModeWithFilter + 64
[... cut ...]
__FONT_DATA                          4K        1 
__LINKEDIT                       391.2M        6 
__OBJC_RO                         32.3M        1 
__OBJC_RW                         1908K        2 
__TEXT                           188.7M      284 
__UNICODE                          564K        1 
mapped file                       49.5M       14 
shared memory                      632K       13 
===========                     =======  ======= 
TOTAL                            910.2M      691 `

ElBalsero, 2021-02-16:
Same here with my Bamboo Fun CTE-450. It works all perfect until I touch the wheel for zooming. Then the preferences pane becomes unavailable and I get the beach ball when I try to access it.
If I change the function of the wheel to any other than zoom (for example pane) then this is not happening and everything works fine.

thenickdude, 2021-02-17:
Thanks for the report! I figured out what is causing this now, it was the same bug that was present in ConsumerTouchDriver when performing a gesture like zoom, but I never fixed the same bug in PenTabletDriver. Apparently PenTabletDriver is the one that is responsible for sending these events when using the touch ring.

I'll get this fixed.

thenickdude, 2021-02-17:
I think I've fixed this now, but I can't test it myself because my tablet doesn't have a scroll ring. Can you two please install this and let me know if it works?

https://github.com/thenickdude/wacom-driver-fix/releases/download/patch-9/Install.Wacom.Tablet-5.3.7-6-patched.pkg

After installation, reboot your computer, and I think it should start working even without playing around with the permissions.

thenickdude, 2021-02-17:
Whoops that link should be https://github.com/thenickdude/wacom-driver-fix/releases/download/patch-9/Install-Wacom-Tablet-5.3.7-6-patched.pkg

ElBalsero, 2021-02-17:
I´ve just tried it and now it works. Not hanging anymore when trying to zoom. And it even actually zooms :-)
Thank you very much for this, Nick. You´re a genious!

thenickdude, 2021-02-18:
Thanks for the test, I've released this fixed version now!

commit 02a690f1 (2021-02-18): Fix PostGesture in Bamboo PenTabletDriver (used for scroll ring gestures)

Closes #60
closed 2021-02-18 by thenickdude