thenickdude/wacom-driver-fix #63: Issue with Wacom driver on Adobe Illustrator Version 25

surmvoise, 2021-02-16:
Since Adobe Illustrator version 25 this dialog window pops up everytime I launch Illustrator. Is thre a fix for that?
[image]

thenickdude, 2021-02-16:
Sorry, I'm not able to test this because I don't own Illustrator (and I already used up my Illustrator trial years ago)

thenickdude, 2021-02-18:
In order to investigate this problem I'll either need someone to donate the 37 USD needed to buy a month of Illustrator, or I'll need someone to share me their login credentials for their Adobe Creative Cloud account which has a license of Illustrator in it.