thenickdude/wacom-driver-fix #65: Intuos draw ctl-490

ferferc, 2021-03-03:
Hi @thenickdude first of all, thank you for the work you do.
I wanna ask you if some of your wacom patches work for my model wacom Intuos Draw CTL-490 because I installed the one for Intuos 3 but didn't work. I'll hope you can help me, because my mac with Catalina OS didn't recognize my tablet.
Thank you.

thenickdude, 2021-03-03:
The CTL-490 is still supported by Wacom's newest driver, so you should just be able to use their official driver. It isn't supported by my Intuos 3 v6.3.15-3 driver, sorry!

closed 2021-03-03 by thenickdude

sermelipharo, 2026-01-20:
https://cdn.wacom.com/u/productsupport/drivers/mac/professional/releasenotes/Mac_6.4.12-3.html — no longer supported, i wonder if it's possible to implement via yours driver? .10 version has major bug making it impossible to use on 26.1
https://cdn.wacom.com/u/productsupport/drivers/mac/professional/releasenotes/Mac_6.4.10-3.html

deanna, 2026-10-04:
Here to say, for anyone else looking for help: I was able to [uninstall](https://support.wacom.com/hc/en-us/articles/1500006264541-How-do-I-uninstall-manually-and-re-install-the-Wacom-driver-on-macOS-for-a-Pen-Tablet-Pen-Display-or-Pen-Computer) all the current Wacom drivers and install [v6.4.9-2]( https://www.wacom.com/en-us/support/product-support/drivers?_gl=1*1iryylr*_ga*MTc1Nzk0MjM5MS4xNzkwMTcxNDg5*_ga_5XHN22BY8E*czE3OTA4NzE0NzAkbzIkZzEkdDE3OTA4NzQxMTAkajYwJGwwJGgxMzg2OTkzNTU1) to get my CTL-490 working again.

My setup: MBA, M2 2022, macOS 26.6.2.