thenickdude/wacom-driver-fix #67: MAJOR problem

adamesq, 2021-03-15:
Just installed the Bamboo driver fix for the first time on a mid2014 MacBook Pro. All seems fine, but once everything boots up - external keyboard stops working, external drives disconnect. The input from even the trackpad on the computer is intermittent. The Bamboo CTH470 light intermittently blinks. Immediate issues is HOW do I uninstall this thing? Then, what happened?!?!?

thenickdude, 2021-03-15:
The Wacom driver can't cause this, it's more likely that your USB cable is broken/damaged (especially at the port end where it plugs into the tablet) and it's causing the whole USB controller to shut down. 

The Wacom driver can't cause this because it's just a regular app running with no special permissions, and just gets given mouse input events from macOS to process. But you're seeing a fault with the whole USB controller, which Wacom doesn't interact with.

adamesq, 2021-03-15:
Thank you so much for the quick reply! I'm trying to get more info on what's happening. The problem doesn't occur until the CTH470 is plugged in. I just restarted, plugged the same cord into a CTL-471 and was asked for input permission. After giving that, the CTL471 worked perfectly. I unplugged it and plugged the same wire into the CTH471 and everything went haywire again. Before installing your driver (before finding it today), I tried the CTH-470 with the Wacom driver. The CTH responded to touch input and recognized the pen, but the Mac didn't respond to pen input (same pen works on CTL471.

Looking in my S&P->Accessibility panel now, I see a PenTabletDriver.app that was installed after I gave the CTL471 permission. Input Monitoring shows PenTabletDriver.app, which I believe was installed after your plugin. I'm not sure if there's a conflict from leftover Bamboo drivers.

thenickdude, 2021-03-15:
Can you open up the "Console" app, and on the left column select "system.log". Copy and paste the log entries that appear on the right side into an email to me at n.sherlock@gmail.com, I'll see if I can spot the trigger for the errors (this data might include your username or apple email address).

adamesq, 2021-03-15:
Will do! 
More notes: I removed all traces of anything Wacom I could find and restarted. After plugging in the CTH470 was back to what I had earlier. The blue light was on steady and turned white when the pen was near. I put the pen down, installed your driver and immediately after the complete message popped up, the external keyboard stopped working, the light on the tablet started blinking and external drives started unmounting. My next thought is that I'm overdrawing power from the USB. I'm gong to plug it in to a powered hub and see what happens.

adamesq, 2021-03-15:
I just plugged the tablet into a power hub and the tablet is working. The MBP is even seeing the pen! Issue resolved. Now, I'm going to remove my post on reddit to avoid confusing anyone. This leaves me wondering what benefits I can get from your plugin if the pad tablet is being recognized without even having drivers installed. I'm new with this tablet and don't know what I should be expecting. Lastly, sorry for startling you in any way on this!

thenickdude, 2021-03-15:
Good to hear that's solved! 

With no functioning driver the tablet gets very basic touch support from the operating system (no gestures / multi-finger) and the pen works like a mouse (no absolute positioning / pressure support). So you really need the driver for it to work in a useful way.

closed 2021-03-15 by thenickdude