thenickdude/wacom-driver-fix #68: Correct path in Manual Instructions

joshua-rodgers, 2021-04-10:
in the manual instructions, in the section for touch support fix, the path:

/Library/Application Support/Tablet/PenTabletDriver.app/Contents/Resources/ConsumerTouchDriver.app/Contents/MacOS

should be:

/Library/Application Support/Tablet/ConsumerTouchDriver.app/Contents/MacOS

Not sure if this is the case on other MacOS versions, but it was for Catalina 10.15.7.

Thanks so much for this fix!

thenickdude, 2021-04-10:
Ahh thanks for that, the path used to be correct but then I actually reorganised the driver itself to fix a problem on Big Sur. I'll update it.

thenickdude, 2021-04-11:
Actually it looks like you used my automatic installer, and then followed the manual directions on top of that? The manual instructions are only if you use the original Wacom installer and want to add my patch on top. If you use my automatic installer you don't have to follow the manual directions at all.

joshua-rodgers, 2021-04-11:
Ah. Yes, thats correct. I ran the installer and connected my tablet to test it out, but it didn't work so i went for the manual method. While i did have the Wacom driver installed, I didn't consider that running your installer might have changed the directory structure. Sorry about that! Thanks for checking it out, this fix was super helpful!

closed 2021-04-11 by thenickdude