thenickdude/wacom-driver-fix #69: (CTF-430) Tablet works but won't accept any button presses

futuraprime, 2021-04-16:
I've installed your patched driver on Big Sur, and it clearly works much better than the official drivers—the cursor now maps across the screen as you'd expect from a tablet, for example.

However, the computer doesn't register any input from the tablet _other than_ positioning. I can move the cursor around with the stylus, but tapping it or pressing the buttons on it does nothing. Any idea what might be preventing the computer from registering the pen input?

thenickdude, 2021-04-16:
In Wacom settings, hold the option/alt key on your keyboard and click the About button, it should open up a diagnostics screen. Check and see if the diagnostics screen is able to report the pen pressure and button events changing.

futuraprime, 2021-04-19:
It picks up x and y coordinates, but switches resolutely remains a bunch of zeroes. Pressure occasionally goes to 1. 
[image]

thenickdude, 2021-04-19:
Do you know if this pen has ever worked on this tablet? Did you get the pen separately from the tablet?

thenickdude, 2021-04-19:
I believe the compatible pen that comes with this tablet is the FP-500, which looks like this:

[image]

futuraprime, 2021-04-19:
I don't know. I purchased both together, from a used electronics shop. They are usually pretty good about confirming their merchandise works, but they may have missed something here.

It does look like the right pen, though.
[image]

futuraprime, 2021-04-19:
FWIW, I am using the pen through a hub, which they recommend against (my laptop has only USB-C ports, so no choice there really).

thenickdude, 2021-04-19:
That looks like the correct pen, yeah. 

I wouldn't expect the hub to be the problem, because I'd expect the pen x,y coordinates to not move either if that was the case. If you ever get [a little USB-C to USB-A converter dongle](https://www.amazon.com/AmazonBasics-Type-C-Gen1-Female-Adapter/dp/B01GGKYYT0/) you can eliminate the hub to test that out (they're completely passive devices, so they don't mess with the signal). I wouldn't bother buying one of those to try to fix this though, unless you think it'll be useful for other things too.

The only other thing I can suggest is to remove the Input Monitoring permission using the minus button (but don't bother removing the Accessibility one), then reboot your Mac, and re-tick the Input Monitoring entry that reappears for you when you try to use the tablet. Then reboot one more time and see if the diagnostics situation improves.

futuraprime, 2021-04-24:
I finally booted up an old computer and ran it with the native drivers on
Yosemite, and hit the same problem. So it’s clearly not your drivers
causing the issue. (Happily, the tablet came warrantied, so I should be
able to get a refund.) Thanks for looking into it!

thenickdude, 2021-04-24:
That's a curious fault! Thanks for getting back to me.

closed 2021-04-24 by thenickdude