thenickdude/wacom-driver-fix #70: Intuos 3 + Mac Catalina - Moves  Cursor - No input

Zodroz, 2021-05-25:
Running Mac OS 10.15.7 on late 2012 iMac.

Intuos 3 was working with original WACOM driver, However the WACOM Mouse buttons sometimes would not work or select a button not directly on top of the cursor.

Installed the patched driver. 

Driver loads and Preference pane recognizes tablet. Even remembered a pen, mouse and app settings (but not mapping). Pen Pressure displays correctly.

BUT pen and mouse cannot select objects or depress buttons.

I also manually installed drivers too.

Any ideas?

thenickdude, 2021-05-25:
If you've already tried the instructions for fixing up the Accessibility permission, try using Wacom Utility to back up and then reset your preferences.

Zodroz, 2021-05-26:
Thanks for the advice! 
I had manually trashed the library files, but that wasn't working when doing the installs (tried multiple times).
Using WACOM utility cleared even the "Tablet" folder in the Library. Maybe there were some invisible files lurking in there? :)
After a reboot my pen and mouse would once again interact with objects (Click buttons, move windows, etc)

Ran your patched installer and it behaved pretty good asking for permission to enable the drivers in Accessibility and Input monitoring. Once I did that and rebooted the WACOM driver is behaving as expected.

However, there is an Illustrator alert that I'm getting each time I open it. I'll make a new thread so anyone else can see the issue. Thanks again!

piccolbo, 2022-03-30:
I am having the same problem with an ancient graphire  and a modern m1 mini and catalina. I have used the Wacom utility to wipe the slate clean, cleared prefs then removed software, then installed this. I tried to fix permission but I don't see anything under Accessibility. I see two related entries under Input monitoring,  I ticked both. Could open a new issue if more appropriate.

thenickdude, 2022-03-30:
You may need to reboot before the Accessibility permission appears.

piccolbo, 2022-03-30:
Actually I take it back, I hadn't seen a dialog asking about permissions, after clicking there I enabled the missing permission item and clicking is back. Can close AFAIAC. Sorry for the false alarm.