thenickdude/wacom-driver-fix #74: Keyboard writing random text

locksoft, 2021-09-07:
Hi again,
after installing the driver, this problem started to appear:

https://discussions.apple.com/thread/252809635

Part of the text I'm writing I'm gets repeated and it's very frustrating.
Do you have any fix? :(

Thanks

thenickdude, 2021-09-07:
Which driver version are you using? Also, which version of macOS? 

Is your keyboard wireless or is this the keyboard built into your Mac?

locksoft, 2021-09-08:
Thanks for your answer.

Driver version: Install-Wacom-Tablet-5.3.7-6-patched.pkg

MacOS version: Big Sur 11.5.2

The keyboard is an old Apple wireless one without the numeric keytab (the one with still the eject button, just to understand).

If I disable WacomTouchDriver and PenTabletDriver from Security > Input monitor (I'm translating from Italian, I guess it's a similar name) then the problem is gone. But, of course, the tablet is not recognized anymore.