thenickdude/wacom-driver-fix #75: Bamboo Fun CTE-450 not seen on Big Sur 11.5.2

toddsines, 2021-09-16:
Removed what I thought was *all* of Wacom's preferences. Tried to install 5.3.7-6-patched; it succeeds, but I cannot load the Wacom driver and correctly see the connected devices.

Any ideas?

toddsines, 2021-09-18:
Anyone? Kind of stumped on this.

thenickdude, 2021-09-18:
If you ever had Wacom's newest v6 driver installed at one point, and didn't use Tablet Utility / Wacom Utility to uninstall it, reinstall that new driver to bring back Tablet Utility, so you can use it to do a full removal.

Then reinstall my driver and follow the instructions to reset the permissions

toddsines, 2021-09-18:
That did it.  I thought I got rid of everything, but yes, using Wacom Utility to remove the Wacom driver is essential for this to work.

Thanks so much for your work on this! I hate throwing old technology away! :)

closed 2021-09-18 by thenickdude