thenickdude/wacom-driver-fix #79: OSX Monterey, M1 Mac, Wacom Bamboo MTE450 (5.3.7-6 Patched)

deadstick86, 2022-01-19:
Thank you so much for the drivers they are incredibly useful! I have it working perfectly once the OS has loaded, but for some reason the table doesn't seem to work at the login stage, can you advise what might be going wrong?

thenickdude, 2022-01-19:
Unfortunately this is a limitation of the newest official Wacom driver too! Because the driver runs as a "regular desktop app", it hasn't been started yet when you're at the login screen.

closed 2022-01-19 by thenickdude

deadstick86, 2022-01-19:
Thank you so much for your fast response and thanks once again for creating
the drivers

Adam.

   Adam Luszniak
   Interiors, Architectural & Studio Photography
   ***@***.***
   +44 (0) 7581 745 172
   www.adamluszniak.com <http://www.adamluszniak.com>