thenickdude/wacom-driver-fix #80: "Wacom Tablet" in System Preferences not working properly | MBA M1, Monterey 12.1 / PTZ-630

XAPNTOH86, 2022-01-19:
Hi! Thirst of all, thanks a lot for the work you're doing!

Not the first time using your driver, however after migrating to an M1 mac with the latest OS, I can't configure my PTZ-630 since almost no buttons in the configuration app in System Preferences work. 

"About", "Advanced", "Options", "+/-" (to add and remove apps) – none of them seem to do anything. Reassigning express keys also doesn't work. Nor does the tip only setting for touch strips – since no response from "Advanced" button.

Eraser pressure is not registering either, and after trying this tip pressure stops registering too, until the tablet is reconnected.

Did the clean install according to the manual. Have WacomSpringBoard in Accessibility and WacomSpringBoard and WacomTabletDriver in InputMonitoring. 

Have tried doing the whole process twice by now, but alas, to no avail :(

XAPNTOH86, 2022-01-19:
Suddenly after the driver crashed all by itself with an error message in System Prefernces, while I wasn't even using the tablet, everything started working properly. 

After unplugging and plugging back the tablet – it's back to square 1 though :(

alex-kovac, 2022-02-09:
I can confirm this issue on Mac Mini M1 with Monterey 12.2

I suspect it has to do something with multiple desktops/spaces on Monterey. There was an occasion when dialog boxes would show up on Desktop 1, but not on Desktops 2,3 or 4.

Also, starting Wacom preferences panel on Desktop 1 will have Mapping options intact, while starting on other desktops leads to the situation on the image below.

[image]