thenickdude/wacom-driver-fix #82: MTE-450 Bamboo Not Working on macOS Monterey (12.1)

rivaldophilip, 2022-01-25:
Situation: 
Followed all the installation instructions but preference still says "Waiting for synchronization." I could hover around the using my pen but it's buggy (always in left click so keep highlighting everything). 

Steps tried:
1. Installed patched driver, remove all user preferences via Wacom utility, uninstalled patched driver, reinstall patched driver
2. Wacom does not show up in Input Monitoring privacy settings. Running "launchctl load -w /Library/LaunchAgents/com.wacom.pentablet.plist" did not work but was able to add PenTabletDriver in Input Monitoring. Checking PenTabletDriver in Input Monitoring does not make items show up in Accessibility settings other than com.wacom.IOManager. 

Could not run "launchctl unload /Library/LaunchAgents/com.wacom.pentablet.plist" and "launchctl load -w /Library/LaunchAgents/com.wacom.pentablet.plis."

thenickdude, 2022-01-25:
>Could not run "launchctl unload /Library/LaunchAgents/com.wacom.pentablet.plist" and "launchctl load -w /Library/LaunchAgents/com.wacom.pentablet.plis."

What errors did you get during that process?

rivaldophilip, 2022-01-25:
Load failed: 5: Input/output error
Try running `launchctl bootstrap` as root for richer errors.

thenickdude, 2022-01-25:
Please show the output of both commands

rivaldophilip, 2022-01-25:
The first command did not return anything. Entering the 2nd command after the first command returned the error I mentioned above (Load failed: 5: Input/output error)

thenickdude, 2022-01-25:
You weren't missing the "t" off the end of the filename like you quoted in your first post were you?

rivaldophilip, 2022-01-26:
No, I just reran the two lines (making sure there are no typos) and it didn't work, same error message

thenickdude, 2022-01-26:
Can you open up the "Console" app, and on the left select the Crash Reports page, and see if there are any recent reports with Tablet in the name? If so you can email the contents of those errors to me at n.sherlock@gmail.com

Also please show me the output of running "ls /Library/LaunchAgents/" in the Terminal app

rivaldophilip, 2022-01-26:
Hello Nick,

I tried to rerun the commands again today and it did not show any error
message but nothing happened either (i.e. wacom still does not show up in
accessibility settings). Attached are the errors from yesterday, and
launchld.log from the console while the commands are running in Terminal.

Regards,
Rivaldo Philip <https://www.linkedin.com/in/rivaldophilip/>

-------------------------------------Translated Report (Full Report Below)-------------------------------------Process:               WacomTabletDriver [1682]Path:                  /Applications/Wacom Tablet.localized/*/WacomTabletDriver.app/Contents/MacOS/WacomTabletDriverIdentifier:            com.wacom.wacomtabletVersion:               Wacom Tablet 6.3.44-2 (6.3.44f2)Code Type:             X86-64 (Native)Parent Process:        launchd [1]User ID:               501Date/Time:             2022-01-25 15:22:19.0146 +0000OS Version:            macOS 12.1 (21C52)Report Version:        12Bridge OS Version:     6.1 (19P647)Anonymous UUID:        3273BE20-9248-B726-331C-1C7806653C66Time Awake Since Boot: 110 secondsSystem Integrity Protection: enabledCrashed Thread:        0Exception Type:        EXC_CRASH (SIGABRT)Exception Codes:       0x0000000000000000, 0x0000000000000000Exception Note:        EXC_CORPSE_NOTIFYTermination Reason:    Namespace DYLD, Code 1 Library missingLibrary not loaded: /Library/Frameworks/WacomMultiTouch.framework/Versions/A/WacomMultiTouchReferenced from: /Applications/Wacom Tablet.localized/*/WacomTabletDriver.app/Contents/MacOS/WacomTabletDriverReason: tried: '/Library/Frameworks/WacomMultiTouch.framework/Versions/A/WacomMultiTouch' (code signature in <1407452C-C6DC-3126-8132-CBCEB9E7065F> '/Library/Frameworks/WacomMultiTouch.framework/Versions/A/WacomMultiTouch' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs), '/System/Library/Frameworks/WacomMultiTouch.framework/Versions/A/WacomMultiTouch' (no such file)(terminated at launch; ignore backtrace)Application Specific Information:Library not loaded: /Library/Frameworks/WacomMultiTouch.framework/Versions/A/WacomMultiTouchReferenced from: /Applications/Wacom Tablet.localized/.Tablet/WacomTabletDriver.app/Contents/MacOS/WacomTabletDriverReason: tried: '/Library/Frameworks/WacomMultiTouch.framework/Versions/A/WacomMultiTouch' (code signature in <1407452C-C6DC-3126-8132-CBCEB9E7065F> '/Library/Frameworks/WacomMultiTouch.framework/Versions/A/WacomMultiTouch' not valid for use in process: mapping process and mapped file (non-platform) have different Team IDs), '/System/Library/Frameworks/WacomMultiTouch.framework/Versions/A/WacomMultiTouch' (no such file)Thread 0 Crashed:0   dyld                          	       0x1099190ce __abort_with_payload + 101   dyld                          	       0x10992f1cf abort_with_payload_wrapper_internal + 802   dyld                          	       0x10992f201 abort_with_payload + 93   dyld                          	       0x1098dd613 dyld4::halt(char const*) + 3754   dyld                          	       0x1098d955d dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 39555   dyld                          	       0x1098d84b4 start + 388Thread 0 crashed with X86 Thread State (64-bit):  rax: 0x0000000002000209  rbx: 0x0000000000000000  rcx: 0x00007ff7b8c87618  rdx: 0x00007ff7b8c87a80  rdi: 0x0000000000000006  rsi: 0x0000000000000001  rbp: 0x00007ff7b8c87660  rsp: 0x00007ff7b8c87618   r8: 0x00007ff7b8c87680   r9: 0x0000000000000000  r10: 0x00000000000000c1  r11: 0x0000000000000246  r12: 0x00000000000000c1  r13: 0x00007ff7b8c87a80  r14: 0x0000000000000001  r15: 0x0000000000000006  rip: 0x00000001099190ce  rfl: 0x0000000000000246  cr2: 0x00000001098dfe3c  Logical CPU:     0Error Code:      0x02000209 Trap Number:     133Binary Images:       0x1098d3000 -        0x10993efff dyld (*) <cef5a27a-d50b-3020-af03-1734b19bc8c5> /usr/lib/dyldExternal Modification Summary:  Calls made by other processes targeting this process:    task_for_pid: 0    thread_create: 0    thread_set_state: 0  Calls made by this process:    task_for_pid: 0    thread_create: 0    thread_set_state: 0  Calls made by all processes on this machine:    task_for_pid: 0    thread_create: 0    thread_set_state: 0VM Region Summary:ReadOnly portion of Libraries: Total=15.0M resident=0K(0%) swapped_out_or_unallocated=15.0M(100%)Writable regions: Total=9268K written=0K(0%) resident=0K(0%) swapped_out=0K(0%) unallocated=9268K(100%)                                VIRTUAL   REGION REGION TYPE                        SIZE    COUNT (non-coalesced) ===========                     =======  ======= STACK GUARD                       56.0M        1 Stack                             8192K        1 VM_ALLOCATE                         12K        3 __DATA                             416K        3 __DATA_CONST                        80K        1 __LINKEDIT                        7984K        4 __TEXT                            7424K        2 dyld private memory               1024K        1 ===========                     =======  ======= TOTAL                             80.5M       16 -----------Full Report-----------{"app_name":"WacomTabletDriver","timestamp":"2022-01-25 15:22:19.00 +0000","app_version":"Wacom Tablet 6.3.44-2","slice_uuid":"5fe48cb1-15cc-3641-953a-fc51d07fe51d","build_version":"6.3.44f2","platform":1,"bundleID":"com.wacom [... cut ...] espawn out by 2 seconds.2022-01-26 11:04:58.303950 (gui/501/com.wacom.pentablet) <Notice>: internal event: WILL_SPAWN, code = 02022-01-26 11:04:58.303953 (gui/501/com.wacom.pentablet) <Notice>: service state: spawn scheduled2022-01-26 11:04:58.303954 (gui/501/com.wacom.pentablet) <Notice>: service throttled by 2 seconds2022-01-26 11:04:58.787372 (system) <Warning>: failed lookup: name = UBF8T346G9.com.microsoft.dlp.net_service, flags = 0x8, requestor = netext[357], error = 3: No such process

thenickdude, 2022-01-26:
Ahh that's an interesting one! I think this is due to a file leftover from the too-new Wacom driver.

In Finder click Go > Go to Folder and enter:

    /Library/Frameworks/

In that folder delete the "WacomMultiTouch.framework" you see there, then reboot and see if everything starts working.

thenickdude, 2022-01-26:
And then reinstall my driver I think

rivaldophilip, 2022-01-26:
Tried it (currently removed all user preferences and tablet software). Now I am back to situation where opening the settings lead me to "Waiting for synchronization" and no Wacom software shows up in Input Monitoring and "com.wacom.IOManager"shows in Accessibility (this is already enabled so I guess not uninstalled via Wacom Utility)?

rivaldophilip, 2022-01-26:
But I have repeated this loop a few times yesterday, which is covered in the steps I have tried

thenickdude, 2022-01-26:
Note that there will be two Wacom utilities, one for my driver and one for the new one.

If you only had one then you might have dragged the old Wacom driver to the trash without using the Utility to properly uninstall it.

In that case use whatever Wacom Utilities you have to uninstall everything, then install Wacom's latest driver to bring back their Wacom Utility. Then you can use that to fully uninstall their driver.

Now that you have a clean system you can reinstall mine.

thenickdude, 2022-01-26:
You can verify what drivers you currently have installed by going to Apple menu > About this mac > System report > Software > Applications. In there you can look for PenTabletDriver / IOManager / WacomTabletDriver / ConsumerTouchDriver and so on and check what version number they have. Version 6.xxx is the too-new driver and version 5.xxx is mine.

rivaldophilip, 2022-01-26:
Tried installing the newest Mac driver and uninstalling it via Wacom Utility but it doesn't seem that I can completely uninstall them (Wacom desktop app, utility, etc.) still exists. Installing your patched package while in this situation does not change the situation of nothing shows in input monitoring, 2 commands to make things show up in accessiblity don't work

moiep, 2022-04-26:
Hello @thenickdude, just wanted to say thank you for creating this driver fix! 
I was able to use it to fix my Bamboo CTH-470 and older Mac last year. However since upgrading to M1 Macbook Pro this year and trying to use the same drivers you've created, I am having exactly the same error messages and issues as Rivaldo. Have followed all your instructions and everything in this comment as well. When I launch the "Wacom" settings, I get an error message saying 'There is a problem with your tablet driver. Please reboot your system. If the problem persists reinstall or update the driver.'

thenickdude, 2022-04-26:
Did you ever install Wacom's too-new driver on that new MacBook?

Try using Wacom Utility to delete your preferences and reboot.

moiep, 2022-04-26:
I've only installed your driver on my new computer.

thenickdude, 2022-04-26:
Did it appear in Input Monitoring and/or Accessibility?

Try open up the "Console" app, go to the "Crash Reports" tab, and check for crashes that have Wacom or Tablet in the name.