thenickdude/wacom-driver-fix #85: Intuos 1 (GD-0608) on Mac M1 Monterey: I can move the cursor but I can't click

solares, 2022-02-21:
However, I see pressure in the Wacom preferences when I push the tip to the tablet - so the mechanics are definitely working, the OS just ignores these taps. 

[image]

thenickdude, 2022-02-21:
That'd have to be a problem with the Accessibility permission, try deleting it using the minus button and reboot, then re-tick the entry that reappears for you when you use the pen, and reboot one more time.

solares, 2022-02-21:
Thanks very much! That did it. 

I ticked both com.wacom.IOManager.app and WacomTabletSpringboard.app. 

[image]

Didn't need the last reboot.

closed 2022-02-21 by solares